openwrt/luci #5147: hostapd does not support country code 00

msqacafe, 2021-06-24:
Currently luci allows you to configure the country code to 00 which creates an invalid hostapd configuration.

hostapd was modified to only allow valid alpha2 country codes.
https://w1.fi/cgit/hostap/commit/?id=e149051ced781ef7b29ae47e98c6d06e4e666f68

openwrt tag: v21.02.0-rc3

dibdot, 2021-06-27:
Not a LuCI issue, the countrylist will be retrieved from iwinfo, e.g. https://git.openwrt.org/?p=project/iwinfo.git;a=blob;f=iwinfo_lib.c;h=70b080cedfc07be6b001f7084ad63ffe69d92f3f;hb=refs/heads/master#l82

nwidger, 2021-06-27:
@dibdot Is https://bugs.openwrt.org/ the correct place to report this issue?

dibdot, 2021-06-27:
yep, that's the correct place for all core components.

jow-, 2021-06-30:
The hostapd commit appears to conflict with https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt#n12 - so far `00` used to be a valid code for selecting the world regulatory domain.

Anyhow, I'll close this here as it needs to be fixed in libiwinfo, but I am not 100% sure if not allowing `00` is correct.

closed 2021-06-30 by jow-