openwrt/luci #5148: Can't login via luci web interface after upgrading to newest 21.02-rc

msmafra, 2021-06-24:
Can't login via LUCI, but still working via ssh. No error is shown in the web interface, but in the logs.

```
❯ cat /etc/os-release
NAME="OpenWrt"
VERSION="21.02.0-rc3"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 21.02.0-rc3"
VERSION_ID="21.02.0-rc3"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r16172-2aba3e9784"
OPENWRT_BOARD="ramips/mt7620"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 21.02.0-rc3 r16172-2aba3e9784"

❯ pkg list-installed | grep luci
liblucihttp-lua - 2021-06-11-3dc89af4-1
liblucihttp0 - 2021-06-11-3dc89af4-1
luci - git-20.074.84698-ead5e81
luci-app-firewall - git-21.163.62626-9660eed
luci-app-opkg - git-21.079.58598-6639e31
luci-base - git-21.163.64918-ba57ec5
luci-lib-base - git-20.232.39649-1f6dc29
luci-lib-ip - git-20.250.76529-62505bd
luci-lib-jsonc - git-19.317.29469-8da8f38
luci-lib-nixio - git-20.234.06894-c4a4e43
luci-mod-admin-full - git-19.253.48496-3f93650
luci-mod-network - git-21.165.25599-2778366
luci-mod-status - git-21.163.64918-6c6559a
luci-mod-system - git-21.163.64918-ca66b63
luci-proto-ipv6 - git-21.148.49484-14511e5
luci-proto-ppp - git-21.163.64918-6c6559a
luci-ssl - git-20.244.36115-e10f954
luci-theme-bootstrap - git-21.164.71418-bd36169
rpcd-mod-luci - 20210614

❯ ps ww | grep rpcd
  957 root      2112 S    /sbin/rpcd -s /var/run/ubus/ubus.sock -t 30
 2858 root      1252 S    grep rpcd

❯ lua -lluci.util -lubus -e 'luci.util.dumptable(ubus.connect():c
all("system", "board", {}))'
board_name	asus,rt-ac51u
hostname	nostromo
release	table: 0x172f470
	distribution	OpenWrt
	revision	r16172-2aba3e9784
	version	21.02.0-rc3
	target	ramips/mt7620
	description	OpenWrt 21.02.0-rc3 r16172-2aba3e9784
model	Asus RT-AC51U
kernel	5.4.124
system	MediaTek MT7620A ver:2 eco:6

❯ logread
...
Thu Jun 24 14:22:26 2021 daemon.err uhttpd[2311]: luci: accepted login on / for root from 141.12.0.238
...
```

Myndex, 2021-06-25:
Manually appending "admin" at the end of the URL so that it's 

    .../cgi-bin/luci/admin
will allow logging into the main Luci interface.

However, associated with this bug is that the Rosy file server is exposed without logging in, which is a concern...

Acris, 2021-06-25:
I have the same problem, if I set the password for the root user via ssh, I can't log in via luci, it says "Invalid username and/or password! Please try again."

By the way, after the installation is done, I can't set the password through luci either when I haven't set the password yet.

Acris, 2021-06-25:
> Manually appending "admin" at the end of the URL so that it's
> 
> ```
> .../cgi-bin/luci/admin
> ```
> 
> will allow logging into the main Luci interface.
> 
> However, associated with this bug is that the Rosy file server is exposed without logging in, which is a concern...

Can't login even if I use `.../cgi-bin/luci/admin`...

Acris, 2021-06-26:
Finally, I solved the problem.

I recompiled the image and the problem went away, but I didn't troubleshoot what software or configuration was causing the problem.

msmafra, 2021-06-28:
> Manually appending "admin" at the end of the URL so that it's
> 
> ```
> .../cgi-bin/luci/admin
> ```
> 
> will allow logging into the main Luci interface.
> 
> However, associated with this bug is that the Rosy file server is exposed without logging in, which is a concern...

Thanks! Sorry for the delayed answer. Even that way it did not work. I use Brave, most of the time.  And I tried in Firefox and Chromium Freeworld, and it worked.

jow-, 2021-06-30:
See https://github.com/openwrt/luci/issues/5104 for the most likely cause and client side solution.

closed 2021-06-30 by jow-