openwrt/luci #5150: OpenWrt 21.02.0-rc3  does not support scsi-like usb-storage by default?

mmokrejs, 2021-06-26:
Hi,
  I flashed 19.07 router with OpenWrt 21.02.0-rc3 and it took me a few hours to fix it after the upgrade. One of the issues was that the following drivers were missing. Most notably, I could not access my USB flash drive as /dev/sda was missing. Please check teh default kernel modules for https://downloads.openwrt.org/releases/21.02.0-rc3/targets/ipq806x/generic/openwrt-21.02.0-rc3-ipq806x-generic-netgear_r7800-squashfs-sysupgrade.bin . 

I installed the following (with some unrelated modules but maybe also worth for inclusion):

```
kmod-fs-ext4 kmod-fs-exfat kmod-fuse kmod-hid kmod-lib-lz4 kmod-lib-lzo kmod-lib-zlib-deflate kmod-lib-zlib-inflate kmod-scsi-core kmod-scsi-generic kmod-usb-storage kmod-usb-storage-uas kmod-usb-uhci kmod-usb3
```

wulfy23, 2021-06-26:
this is tairly atypical and has been the case for several releases... ( and it's not luci specific )

default images contain what is needed for the specific hardware to reduce bloat and maintain compatibility across several boards factoring the lowest and most common denominators...

building your own images provides you with an opportunity to include additional packages...

submitting a PR to the main repo provides an opportunity to alter default package sets.

mmokrejs, 2021-06-26:
I will not make a PR for this as I am not familiar with the development of OpenWRT. Would you reming at least pointing to `mod-scsi-core kmod-scsi-generic kmod-usb-storage`? Thank you.

wulfy23, 2021-06-27:
apologies... I don't understand the question...

as mentioned... it's not related to luci... they are deliberately not included

if you think this is a bug or regression you can report a bug at 'openwrt flyspray'

if you think this is required and are unable to offer any solution... you can email the developer mailing list

mmokrejs, 2021-06-27:
Ah, now I understand. I did not want to report the bug under "luci" but under "packages". Sorry for the noise.

closed 2021-06-27 by mmokrejs