openwrt/luci #5163: luci-app-statistics: hangs with "loading data" when has lot of data

kpoman, 2021-07-02:
Hello,
I had my luci-app-statistics running well for a couple of weeks, but now, I a unable to see anything. When I try to see statistics it just stays showing "loading data" message, but nothing appears. After some seconds or minutes, anything that I click on luci brings me to the login page, as if I was disconnected from my current session. This happens to me for a powerful router (rpi4) or for a simple one (tpllink c7 + usb), on recent snapshot.
Just for the record, here are my stats as of now:
```
root@rpi4:~# du /opt/stats/
76	/opt/stats/rpi4/ping
44	/opt/stats/rpi4/iwinfo-wlan0
52	/opt/stats/rpi4/memory
24	/opt/stats/rpi4/load
32	/opt/stats/rpi4/ipstatistics-all
68	/opt/stats/rpi4/interface-br-lan
12	/opt/stats/rpi4/thermal-thermal_zone0
28	/opt/stats/rpi4/sqm-eth1
280	/opt/stats/rpi4/sqmcake-eth1
68	/opt/stats/rpi4/cpu-0
68	/opt/stats/rpi4/cpu-1
68	/opt/stats/rpi4/cpu-2
68	/opt/stats/rpi4/cpu-3
92	/opt/stats/rpi4/tcpconns-80-local
92	/opt/stats/rpi4/tcpconns-22-local
12	/opt/stats/rpi4/uptime
92	/opt/stats/rpi4/disk-mmcblk0p2
58772	/opt/stats/rpi4/netlink-eth1
28	/opt/stats/rpi4/conntrack
59980	/opt/stats/rpi4
59984	/opt/stats/

```

hnyman, 2021-07-02:
Does the problem occur, if you use the default data directory is /tmp instead of /opt ?

Note that the data directory path needs to be readable by everyone.
/opt and /opt/stats

jow-, 2021-07-02:
If you're thrown to the login page then rpcd most likely terminated.

kpoman, 2021-07-04:
> If you're thrown to the login page then rpcd most likely terminated. Please provide the complete output of `find /opt/stats/rpi4`
> 
> I guess the file listing reply is too large, crashing rpcd.

Right, something really weird here:

/opt/stats/rpi4/
/opt/stats/rpi4/netlink-eth1
/opt/stats/rpi4/netlink-eth1/if_tx_dropped-class-cake-800d:4ba.rrd
/opt/stats/rpi4/netlink-eth1/if_tx_dropped-class-cake-8019:4c6.rrd
/opt/stats/rpi4/netlink-eth1/if_tx_dropped-class-cake-8009:ba2.rrd
.... 7396 lines of similar stuff ....
/opt/stats/rpi4/netlink-eth1/if_tx_dropped-class-cake-800d:62f.rrd
/opt/stats/rpi4/netlink-eth1/if_tx_dropped-class-cake-8009:42e.rrd
/opt/stats/rpi4/netlink-eth1/if_tx_dropped-class-cake-8009:7f4.rrd
/opt/stats/rpi4/cpu-2
/opt/stats/rpi4/cpu-2/percent-wait.rrd
/opt/stats/rpi4/cpu-2/percent-system.rrd
/opt/stats/rpi4/cpu-2/percent-interrupt.rrd
/opt/stats/rpi4/cpu-2/percent-idle.rrd
/opt/stats/rpi4/cpu-2/percent-steal.rrd
/opt/stats/rpi4/cpu-2/percent-user.rrd
/opt/stats/rpi4/cpu-2/percent-softirq.rrd
/opt/stats/rpi4/cpu-2/percent-nice.rrd
/opt/stats/rpi4/interface-br-lan
/opt/stats/rpi4/interface-br-lan/if_packets.rrd
/opt/stats/rpi4/interface-br-lan/if_octets.rrd
/opt/stats/rpi4/interface-br-lan/if_errors.rrd
/opt/stats/rpi4/interface-br-lan/if_dropped.rrd
/opt/stats/rpi4/thermal-thermal_zone0
/opt/stats/rpi4/thermal-thermal_zone0/temperature.rrd
/opt/stats/rpi4/load
/opt/stats/rpi4/load/load.rrd
/opt/stats/rpi4/tcpconns-80-local
/opt/stats/rpi4/tcpconns-80-local/tcp_connections-FIN_WAIT2.rrd
/opt/stats/rpi4/tcpconns-80-local/tcp_connections-CLOSED.rrd
/opt/stats/rpi4/tcpconns-80-local/tcp_connections-CLOSING.rrd
/opt/stats/rpi4/tcpconns-80-local/tcp_connections-LISTEN.rrd
/opt/stats/rpi4/tcpconns-80-local/tcp_connections-SYN_RECV.rrd
/opt/stats/rpi4/tcpconns-80-local/tcp_connections-TIME_WAIT.rrd
/opt/stats/rpi4/tcpconns-80-local/tcp_connections-ESTABLISHED.rrd
/opt/stats/rpi4/tcpconns-80-local/tcp_connections-FIN_WAIT1.rrd
/opt/stats/rpi4/tcpconns-80-local/tcp_connections-SYN_SENT.rrd
/opt/stats/rpi4/tcpconns-80-local/tcp_connections-CLOSE_WAIT.rrd
/opt/stats/rpi4/tcpconns-80-local/tcp_connections-LAST_ACK.rrd
/opt/stats/rpi4/sqm-eth1
/opt/stats/rpi4/sqm-eth1/qdisc_bytes.rrd
/opt/stats/rpi4/sqm-eth1/qdisc_backlog.rrd
/opt/stats/rpi4/sqm-eth1/qdisc_drops.rrd
/opt/stats/rpi4/uptime
/opt/stats/rpi4/uptime/uptime.rrd
/opt/stats/rpi4/ping
/opt/stats/rpi4/ping/ping_droprate-192.168.10.1.rrd
/opt/stats/rpi4/ping/ping-127.0.0.1.rrd
/opt/stats/rpi4/ping/ping_stddev-192.168.10.1.rrd
/opt/stats/rpi4/ping/ping_droprate-8.8.8.8.rrd
/opt/stats/rpi4/ping/ping_stddev-127.0.0.1.rrd
/opt/stats/rpi4/ping/ping_stddev-8.8.8.8.rrd
/opt/stats/rpi4/ping/ping-192.168.10.1.rrd
/opt/stats/rpi4/ping/ping-8.8.8.8.rrd
/opt/stats/rpi4/ping/ping_droprate-127.0.0.1.rrd
/opt/stats/rpi4/iwinfo-wlan0
/opt/stats/rpi4/iwinfo-wlan0/stations.rrd
/opt/stats/rpi4/iwinfo-wlan0/bitrate.rrd
/opt/stats/rpi4/iwinfo-wlan0/signal_noise.rrd
/opt/stats/rpi4/iwinfo-wlan0/signal_power.rrd
/opt/stats/rpi4/iwinfo-wlan0/signal_quality.rrd
/opt/stats/rpi4/memory
/opt/stats/rpi4/memory/memory-buffered.rrd
/opt/stats/rpi4/memory/memory-slab_unrecl.rrd
[... 50 lines cut ...]
/opt/stats/rpi4/tcpconns-22-local/tcp_connections-CLOSED.rrd
/opt/stats/rpi4/tcpconns-22-local/tcp_connections-CLOSING.rrd
/opt/stats/rpi4/tcpconns-22-local/tcp_connections-LISTEN.rrd
/opt/stats/rpi4/tcpconns-22-local/tcp_connections-SYN_RECV.rrd
/opt/stats/rpi4/tcpconns-22-local/tcp_connections-TIME_WAIT.rrd
/opt/stats/rpi4/tcpconns-22-local/tcp_connections-ESTABLISHED.rrd
/opt/stats/rpi4/tcpconns-22-local/tcp_connections-FIN_WAIT1.rrd
/opt/stats/rpi4/tcpconns-22-local/tcp_connections-SYN_SENT.rrd
/opt/stats/rpi4/tcpconns-22-local/tcp_connections-CLOSE_WAIT.rrd
/opt/stats/rpi4/tcpconns-22-local/tcp_connections-LAST_ACK.rrd
/opt/stats/rpi4/ipstatistics-all
/opt/stats/rpi4/ipstatistics-all/ip_stats_octets.rrd
/opt/stats/rpi4/cpu-0
/opt/stats/rpi4/cpu-0/percent-wait.rrd
/opt/stats/rpi4/cpu-0/percent-system.rrd
/opt/stats/rpi4/cpu-0/percent-interrupt.rrd
/opt/stats/rpi4/cpu-0/percent-idle.rrd
/opt/stats/rpi4/cpu-0/percent-steal.rrd
/opt/stats/rpi4/cpu-0/percent-user.rrd
/opt/stats/rpi4/cpu-0/percent-softirq.rrd
/opt/stats/rpi4/cpu-0/percent-nice.rrd
/opt/stats/rpi4/conntrack
/opt/stats/rpi4/conntrack/conntrack.rrd
/opt/stats/rpi4/conntrack/percent-used.rrd
/opt/stats/rpi4/conntrack/conntrack-max.rrd

hnyman, 2021-07-05:
You better disable the netlink stats, if you have lots of cake-related classes/streams/whatever.

```
/opt/stats/rpi4/netlink-eth1/if_tx_dropped-class-cake-8019:4c6.rrd
/opt/stats/rpi4/netlink-eth1/if_tx_dropped-class-cake-8009:ba2.rrd

!!!! .... 7396 lines of similar stuff .... !!!!!

/opt/stats/rpi4/netlink-eth1/if_tx_dropped-class-cake-800d:62f.rrd
/opt/stats/rpi4/netlink-eth1/if_tx_dropped-class-cake-8009:42e.rrd
```

(or configure the netlink plugin to ignore those?   I think that you can configure the netlink plugin by various ways)

jow-, 2021-07-05:
Yes, to work around the problem for now you need to reduce the amount of rrd files until the large reply crash is solved

kpoman, 2021-07-06:
Ok, I confirm that after stopping that collectd plugin, then deleting the local rrd files on that folder /opt/stats/rpi4/netlink-eth1/, it started working again. So maybe should I post this as a bug of that collectd plugin ? Or as a crash of rpcd ? Do you think I should report it or is it a known limitation ?

hnyman, 2021-07-06:
I feel that having over 8000 rrd files (with long names) there, causing the file list size to grow to  > 512 kB, is not an expected use case.

Jow knows better, but it sounds like rpcd limitation, or lua limitation? (But it is not a collects problem as data is collected normally)

I don't  think that the separate stats for 8000+ randomly named cake traffic classes (flows?) are really meaningful. Or do you have an actual use case for those stats?

kpoman, 2021-07-06:
It was auto-built by having netlink + sqm stats activated, which may be pretty common. Maybe netlink collectd plugin maintainers should be alerted ?