openwrt/luci #5179: luci-app-acl do not work on 21.02-RC3

erdoukki, 2021-07-09:
Environment: (mvebu, espressobin-ultra, OpenWrt 21-RC3)

https://forum.openwrt.org/t/changing-the-username-root/100678/14

> I have tried on 21.02-RC3, not working, (error on logging).
> I tried the 3 methods, plain password, hash password and shadow (with installing all shadow packages and adding a user...)
> No success...

jow-, 2021-07-09:
Define what is not working. Account management page crashing, /etc/config/rpcd not updated? Login not working? Login working but blank page displayed?

Also what „error on logging“ exactly? Please paste the relevant lines

erdoukki, 2021-07-09:
I will retest ASAP, but it was a message in the login page of unknown login... (something like this, will correct the report soon)

jow-, 2021-07-09:
You might need to manually restart/etc/init.d/rpcd after adding the account.

erdoukki, 2021-07-09:
```
Invalid username and/or password! Please try again.
```

erdoukki, 2021-07-09:
```
Invalid username and/or password! Please try again.
```

still issued when trying admin login after a full reboot !

jow-, 2021-07-13:
It worked for me on 21.02 rc3 after restarting rpcd on the command line. Was able to log in with the newly added user.

closed 2021-07-13 by jow-

erdoukki, 2021-07-13:
okay for me on 21.02-RC3 with hash password after restarting rpcd service or a manual reboot of OpenWrt.