openwrt/luci #5184: Shedule task not start

mikko70, 2021-07-12:
In sheduled tasks show note "Note: you need to manually restart the cron service if the crontab file was empty before editing."
I have TP-link c7 v2 and i must do manual Cron restart every time i change shedule time or something there.
I hope add automated cron restart every time i press save button. Then this cause smaller amount confused questions with new users like me.

mikko70, 2021-07-12:
[image]

professor-jonny, 2021-07-12:
how it says is how it works.
on boot if the cron is empty the task closes, if not it runs.
rebooting the  device enacles the service you can also start the cron service from ssh by issuing the following command:
`service cron restart`

mikko70, 2021-07-14:
But it not need be  empty, i say every time.

professor-jonny, 2021-07-14:
So you want the wording to be updated if it is not empty and running it will say somthing else ?

mikko70, 2021-07-15:
Reproducce problem.

Step 1.
Put in shedule

00 10 * * *  /sbin/wifi down
02 10 * * *  /sbin/wifi up; wifi

/etc/init.d/cron restart
It working, wifi go on 10am 00 and off 10am 02

Step2.
Put in shedule

05 10 * * *  /sbin/wifi down
07 10 * * *  /sbin/wifi up; wifi

It not working

Step3.
Put in shedule

10 10 * * *  /sbin/wifi down
12 10 * * *  /sbin/wifi up; wifi

/etc/init.d/cron restart
It working, wifi go on 10am 10 and off 10am 12

Step4.
Put in shedule

14 10 * * *  /sbin/wifi down
16 10 * * *  /sbin/wifi up; wifi

/etc/init.d/cron restart
It working, wifi go on 10am 14 and off 10am 16

Step5.
Put in shedule

16 10 * * *  /sbin/wifi down
18 10 * * *  /sbin/wifi up; wifi

/etc/init.d/cron restart
It working, wifi go on 10am 16and off 10am 18

Step6.
Put in shedule

20 10 * * *  /sbin/wifi down
07 10 * * *  /sbin/wifi up; wifi

It not working

professor-jonny, 2021-07-15:
That wont work f devices are associated with the connection at the time of shutdown you are using the wrong command.

Instead use below like I said above and it will force off the wifi interface even if clients are connected
but you will need to customise your wifi name by using the command `uci show wireless`

this is my wirless config and i want to turn off the kids5g interface so I use wireless.wifinet 2 interface 

```
wireless.radio0=wifi-device
wireless.default_radio0.ssid='Admin2.5g'
wireless.default_radio1.ssid='Admin5g'
wireless.wifinet2=wifi-iface
wireless.wifinet2.device='radio1'
wireless.wifinet2.ssid='Kids5g'
wireless.wifinet2.network='KidsLan'
wireless.wifinet3.device='radio0'
wireless.wifinet3.ssid='Kids2.5g'
wireless.wifinet3.network='KidsLan'
```

The below pasting in my interface name it will turn off at 11pm and 11:10 pm:
and turn back on at 7 and 7:10 am

the reason i have pasted in 0 and 10 minutes if for some reason the kids get smart and turn off the router at 11pm it will miss the cron job and it will then work again 10 minutes later to fool the kids :-)

```
0/10 23 * * * uci set wireless.wifinet2.disabled=1; uci commit; wifi
0/10 7 * * * uci set wireless.wifinet2.disabled=0; uci commit; wifi

```

jow-, 2021-07-15:
Don‘t add leading zeroes to your tine specs. Also not LuCi related.

professor-jonny, 2021-07-15:
I agree this is not luci related but he does I think make a point that the note does not change if the service is running.
Can luci interface include a command start the cron service on pressing save or maybe something to start or show the status of cron?

commit 8d4ac60f (2021-07-15): luci-mod-system: reload crond upon saving crontab

Fixes: #5184
commit de8bc46a (2021-07-15): luci-mod-system: reload crond upon saving crontab

Fixes: #5184

(cherry picked from commit 8d4ac60f7a02e5c8e3d186f4009b4696b098152d)
closed 2021-07-15 by jow-