openwrt/luci #5185: luci-app-mwan3: must Flush conntrack table on interface event?

ptpt52, 2021-07-13:
https://github.com/openwrt/luci/blob/b6ca2947e93ddd5d5c4a43b51e75d79f4ce06049/applications/luci-app-mwan3/htdocs/luci-static/resources/view/mwan3/network/interface.js#L255

what if I want to disable flush action?

feckert, 2021-07-13:
I must have overlooked that when I switched from Lua to Javascript.

ptpt52, 2021-08-06:
probably `empty` value should be listed and name as `Disabled`
and set it to be default.

```
diff --git a/applications/luci-app-mwan3/htdocs/luci-static/resources/view/mwan3/network/interface.js b/applications/luci-app-mwan3/htdocs/luci-static/resources/view/mwan3/network/interface.js
index 00cebdfd0b..bb93e3ec86 100644
--- a/applications/luci-app-mwan3/htdocs/luci-static/resources/view/mwan3/network/interface.js
+++ b/applications/luci-app-mwan3/htdocs/luci-static/resources/view/mwan3/network/interface.js
@@ -257,6 +257,8 @@ return view.extend({
 
                o = s.option(form.ListValue, 'flush_conntrack', _('Flush conntrack table'),
                        _('Flush global firewall conntrack table on interface events'));
+               o.default='';
+               o.value('', _('Disabled'));
                o.value('ifup', _('ifup (netifd)'));
                o.value('ifdown', _('ifdown (netifd)'));
                o.value('connected', _('connected (mwan3)'));
```

feckert, 2021-08-12:
Fixed with commit https://github.com/openwrt/luci/commit/57688b9ce824fe54c55158e15f61d7e87fe0dd4f

closed 2021-08-12 by feckert