openwrt/luci #5187: (luci-app-https-dns-proxy) Mullvad Support

x4e, 2021-07-13:
Hello,

It would be great if support could be added for Mullvad's DoH service (https://mullvad.net/en/help/dns-over-https-and-dns-over-tls/).

I believe these are the required configuration files:
```lua
return{
name="Mullvad",
label=_("Mullvad"),
resolver_url="https://doh.mullvad.net/dns-query",
bootstrap_dns="194.242.2.2,193.19.108.2,2a07:e340::2",
help_link="https://mullvad.net/en/help/dns-over-https-and-dns-over-tls/",
help_text="mullvad.net/en/help/dns-over-https-and-dns-over-tls/"
}
```
```lua
return{
name="Mullvad (Ad Blocker)",
label=_("Mullvad (Ad Blocker)"),
resolver_url="https://adblock.doh.mullvad.net/dns-query",
bootstrap_dns="194.242.2.3,193.19.108.3,2a07:e340::3",
help_link="https://mullvad.net/en/help/dns-over-https-and-dns-over-tls/",
help_text="mullvad.net/en/help/dns-over-https-and-dns-over-tls/"
}
```

Also, if you manually set a DoH server (not currently defined by `luci-app-https-dns-proxy`) through `uci`, `luci-app-https-dns-proxy` will display it as being the CIRA (Family) server.

Thank You

stangri, 2021-07-14:
@x4e have you tested if these work? Do you want to send the PR yourself or would you rather I add these on my own?

Outside of adding resolver files I've stopped working on the WebUI until I have the time to rewrite in javascript for client-side rendering as it is the new standard.

x4e, 2021-07-14:
> @x4e have you tested if these work? Do you want to send the PR yourself or would you rather I add these on my own?

I have not tested them and I would prefer not to send the PR myself due to requiring a full name sign-off. I will be able to test it within a couple of days though if that will help.

> Outside of adding resolver files I've stopped working on the WebUI until I have the time to rewrite in javascript for client-side rendering as it is the new standard.

That's a shame, I was already thinking it relied on too much JavaScript!

stangri, 2021-07-16:
[Mullvad DoH servers don't seem to work well with https-dns-proxy](https://github.com/aarond10/https_dns_proxy/issues/123). I've fixed two issues with the files you've posted (help_link_text instead of help_link) and bootstrap servers should be the regular public DNS servers, not DoH servers IPs, if there's a way to make mullvad DoH servers work with the principal package, I'll push the updates.

stangri, 2021-07-16:
@neheb @aparcar @jow- My makefile-fu is not strong enough to modify the [curl Makefile](https://github.com/openwrt/packages/blob/master/net/curl/Makefile) to build a libcurl with openssl variant, can anyone help?

libcurl with wolfssl doesn't support HTTP/2 and some DoH resolvers now require it, so libcurl with openssl would be preferred for https-dns-proxy. 

This is probably not the most appropriate place to bring this up, should I open an issue asking for help in packages?

x4e, 2021-07-16:
Unfortunately I don’t think Mullvad has any public non-DoH DNS servers, so at the moment it would be impossible to use them - at least not without falling back onto another DNS provider for the bootstrapping. Perhaps I could make a PR for https-dns-proxy to allow directly providing the ip of the DoH server instead of a bootstrap DNS from which to resolve it.

SvenKiljan, 2021-12-14:
Mullvad has a public non-DoH DNS server:

https://mullvad.net/nl/blog/2019/2/20/our-public-dns-changing/

stangri, 2021-12-14:
Forgot it was open. Mullvad support has been added a while ago: https://github.com/openwrt/luci/pull/5216 https://github.com/openwrt/luci/pull/5217 https://github.com/openwrt/luci/pull/5218.

closed 2021-12-14 by stangri

stangri, 2022-09-19:
> > Forgot it was open. Mullvad support has been added a while ago: #5216 #5217 #5218.
> 
> Mullvad is no longer in dns instances. Has it been removed?

If I understood you correctly -- no, it's still there, the list may not be alphabetically sorted.

yslupdates, 2022-12-30:
Still unable to see Mullvad in server list. Is there a way to add a custom DNS through luci?

stangri, 2022-12-30:
> Still unable to see Mullvad in server list. Is there a way to add a custom DNS through luci?

You're most likely using outdated `luci-app-https-dns-proxy`.

yslupdates, 2022-12-30:
I'm currently on version 2021-11-22-7 and don't see any updates for it. I'm on Openwrt 22.03.2

What is the latest?

stangri, 2022-12-30:
As evident from the code: https://github.com/openwrt/luci/tree/openwrt-22.03/applications/luci-app-https-dns-proxy/luasrc/https-dns-proxy/providers this version has two mullvad servers:
https://github.com/openwrt/luci/blob/openwrt-22.03/applications/luci-app-https-dns-proxy/luasrc/https-dns-proxy/providers/net.mullvad.doh.adblocker.lua
https://github.com/openwrt/luci/blob/openwrt-22.03/applications/luci-app-https-dns-proxy/luasrc/https-dns-proxy/providers/net.mullvad.doh.lua

If you're on the current version, it may be the browser caching issue.

yslupdates, 2022-12-31:
https-dns-proxy 2022-10-15-10, is this the latest version? The 2021-11-22-7 was for luci, sorry

stangri, 2022-12-31:
My previous comment applies then.

yslupdates, 2022-12-31:
I've used multiple browsers and their respective private modes but still cannot see Mullvad DNS listed.

This is my view, if I can provide any logs or anything else let me know

[image]

stangri, 2022-12-31:
Please post the output of the following commands ran in CLI on router:

```sh
opkg list-installed | grep https
ls -la /usr/lib/lua/luci/https-dns-proxy/providers
```

yslupdates, 2022-12-31:
root@OpenWrt:~# opkg list-installed | grep https
 /usr/lib/lua/luci/https-dns-proxy/providershttps-dns-proxy - 2022-10-15-10
luci-app-https-dns-proxy - 2021-11-22-7
root@OpenWrt:~# ls -la /usr/lib/lua/luci/https-dns-proxy/providers
drwxr-xr-x    2 root     root          6632 Dec 30 00:37 .
drwxr-xr-x    4 root     root           312 Dec 27 23:20 ..
-rw-r--r--    1 root     root           251 Dec 27 23:20 app.tiar.doh.lua
-rw-r--r--    1 root     root           253 Dec 27 23:20 app.tiar.jp.lua
-rw-r--r--    1 root     root           367 Dec 27 23:20 ca.cira.canadianshield.family.lua
-rw-r--r--    1 root     root           370 Dec 27 23:20 ca.cira.canadianshield.private.lua
-rw-r--r--    1 root     root           376 Dec 27 23:20 ca.cira.canadianshield.protected.lua
-rw-r--r--    1 root     root           415 Dec 27 23:20 ch.digitale-gesellschaft.dns.lua
-rw-r--r--    1 root     root           263 Dec 27 23:20 ch.switch.dns.lua
-rw-r--r--    1 root     root           187 Dec 27 23:20 cn.360.doh.lua
-rw-r--r--    1 root     root           273 Dec 27 23:20 cn.rubyfish.dns.lua
-rw-r--r--    1 root     root           239 Dec 27 23:20 co.osxz.dns.lua
-rw-r--r--    1 root     root           297 Dec 27 23:20 com.adguard.dns-family.lua
-rw-r--r--    1 root     root           274 Dec 27 23:20 com.adguard.dns.lua
-rw-r--r--    1 root     root           344 Dec 27 23:20 com.ahadns.blitz.lua
-rw-r--r--    1 root     root           247 Dec 27 23:20 com.alidns.dns.lua
-rw-r--r--    1 root     root           330 Dec 27 23:20 com.blahdns.doh-ch.lua
-rw-r--r--    1 root     root           330 Dec 27 23:20 com.blahdns.doh-de.lua
-rw-r--r--    1 root     root           330 Dec 27 23:20 com.blahdns.doh-fi.lua
-rw-r--r--    1 root     root           330 Dec 27 23:20 com.blahdns.doh-jp.lua
-rw-r--r--    1 root     root           330 Dec 27 23:20 com.blahdns.doh-sg.lua
-rw-r--r--    1 root     root           318 Dec 27 23:20 com.cloudflare-dns.family.lua
-rw-r--r--    1 root     root           301 Dec 27 23:20 com.cloudflare-dns.lua
-rw-r--r--    1 root     root           324 Dec 27 23:20 com.cloudflare-dns.security.lua
-rw-r--r--    1 root     root           264 Dec 27 23:20 com.controld.freedns.family.lua
-rw-r--r--    1 root     root           278 Dec 27 23:20 com.controld.freedns.malware-ads-social.lua
-rw-r--r--    1 root     root           262 Dec 27 23:20 com.controld.freedns.malware-ads.lua
-rw-r--r--    1 root     root           252 Dec 27 23:20 com.controld.freedns.malware.lua
-rw-r--r--    1 root     root           260 Dec 27 23:20 com.controld.freedns.p0.lua
-rw-r--r--    1 root     root           263 Dec 27 23:20 com.controld.freedns.p1.lua
-rw-r--r--    1 root     root           269 Dec 27 23:20 com.controld.freedns.p2.lua
-rw-r--r--    1 root     root           278 Dec 27 23:20 com.controld.freedns.p3.lua
-rw-r--r--    1 root     root           252 Dec 27 23:20 com.controld.freedns.unfiltered.lua
-rw-r--r--    1 root     root           333 Dec 27 23:20 com.decloudus.dns.lua
-rw-r--r--    1 root     root           281 Dec 27 23:20 com.dnsforfamily.dns-doh.lua
-rw-r--r--    1 root     root           282 Dec 27 23:20 com.dnslify.doh.lua
-rw-r--r--    1 root     root           309 Dec 27 23:20 com.opendns.doh.lua
-rw-r--r--    1 root     root           351 Dec 27 23:20 com.opendns.familyshield.doh.lua
-rw-r--r--    1 root     root           253 Dec 27 23:20 com.pumplex.dns.lua
-rw-r--r--    1 root     root           358 Dec 27 23:20 com.rethinkdns.basic.lua
-rw-r--r--    1 root     root           255 Dec 27 23:20 cz.nic.odvr.lua
-rw-r--r--    1 root     root           273 Dec 27 23:20 de.dnsforge.lua
-rw-r--r--    1 root     root           339 Dec 27 23:20 fi.lelux.resolver-eu.lua
-rw-r--r--    1 root     root           180 Dec 27 23:20 google.dns.lua
-rw-r--r--    1 root     root           239 Dec 27 23:20 gr.libredns.doh-ads.lua
-rw-r--r--    1 root     root           232 Dec 27 23:20 gr.libredns.doh.lua
-rw-r--r--    1 root     root           235 Dec 27 23:20 io.nextdns.dns.lua
-rw-r--r--    1 root     root           228 Dec 27 23:20 io.seby.doh-2.lua
-rw-r--r--    1 root     root           335 Dec 27 23:20 jp.iij.dns.public.lua
-rw-r--r--    1 root     root           366 Dec 27 23:20 lu.restena.kaitain.lua
-rw-r--r--    1 root     root           381 Dec 27 23:20 net.ahadns.au.doh.lua
-rw-r--r--    1 root     root           391 Dec 27 23:20 net.ahadns.chi.doh.lua
-rw-r--r--    1 root     root           381 Dec 27 23:20 net.ahadns.es.doh.lua
-rw-r--r--    1 root     root           381 Dec 27 23:20 net.ahadns.in.doh.lua
-rw-r--r--    1 root     root           381 Dec 27 23:20 net.ahadns.it.doh.lua
-rw-r--r--    1 root     root           393 Dec 27 23:20 net.ahadns.la.doh.lua
-rw-r--r--    1 root     root           381 Dec 27 23:20 net.ahadns.nl.doh.lua
-rw-r--r--    1 root     root           381 Dec 27 23:20 net.ahadns.no.doh.lua
-rw-r--r--    1 root     root           390 Dec 27 23:20 net.ahadns.ny.doh.lua
-rw-r--r--    1 root     root           381 Dec 27 23:20 net.ahadns.pl.doh.lua
-rw-r--r--    1 root     root           380 Dec 27 23:20 net.applied-privacy.lua
-rw-r--r--    1 root     root           172 Dec 27 23:20 net.cfiec.dns.lua
-rw-r--r--    1 root     root           345 Dec 27 23:20 net.ffmuc.doh.lua
-rw-r--r--    1 root     root           254 Dec 27 23:20 net.he.ordns.lua
-rw-r--r--    1 root     root           161 Dec 27 23:20 net.idnet.doh.lua
[... cut ...]
-rw-r--r--    1 root     root           316 Dec 27 23:20 org.cleanbrowsing.doh-adult.lua
-rw-r--r--    1 root     root           319 Dec 27 23:20 org.cleanbrowsing.doh-family.lua
-rw-r--r--    1 root     root           325 Dec 27 23:20 org.cleanbrowsing.doh-security.lua
-rw-r--r--    1 root     root           354 Dec 27 23:20 org.snopyta.dns.doh.fi.lua
-rw-r--r--    1 root     root           248 Dec 27 23:20 pub.doh.lua
-rw-r--r--    1 root     root           238 Dec 27 23:20 sb.dns.lua
-rw-r--r--    1 root     root           278 Dec 27 23:20 tw.twnic.dns.lua
root@OpenWrt:~#

stangri, 2022-12-31:
Oh, sorry, forgot there's a piece of code which checks if the installed curl is HTTP/2 compatible and doesn't show incompatible providers. Mullvad is the HTTP/2 only provider.

You need to install `libnghttp2-14` and check that `curl --version` says HTTP2 in Features.

yslupdates, 2022-12-31:
Installing curl did it for me. Thank you so much for taking the time to help