openwrt/luci #5189: Luci wireless status page not showing associations

turboproc, 2021-07-14:
Hi,

using OpenWrt SNAPSHOT r17119-1236cbe30c (that's Master of today) I noticed that the Wireless status page in Luci does not display the associations. A message states "collecting data" Something similar happens on the overall status page with a message "No information available".

Did some investigation and notices that ubus call are made on regular intervals and that ubus responses contain valid data. May be related to unfinished refactoring ?

Sample ubus response:

[{"jsonrpc":"2.0","id":55,"result":[0,{"results":[]}]},{"jsonrpc":"2.0","id":56,"result":[0,{"results":[{"mac":"B8:27:XX:XX:XX:XX","signal":-53,"signal_avg":-53,"noise":-103,"inactive":10,"connected_time":1006,"thr":0,"authorized":true,"authenticated":true,"preamble":"short","wme":true,"mfp":false,"tdls":false,"mesh llid":0,"mesh plid":0,"mesh plink":"","mesh local PS":"","mesh peer PS":"","mesh non-peer PS":"","rx":{"drop_misc":31,"packets":1779879,"bytes":-1559397097,"ht":true,"vht":false,"he":false,"mhz":20,"rate":72200,"mcs":7,"40mhz":false,"short_gi":true},"tx":{"failed":1,"retries":5354,"packets":406680,"bytes":35125484,"ht":true,"vht":false,"he":false,"mhz":20,"rate":72200,"mcs":7,"40mhz":false,"short_gi":true}}]}]}]

jow-, 2021-07-14:
Works here for me. Check your browser debug console for errors.

turboproc, 2021-07-14:
Well, that's the issue: no errors presented on console at all.

jow-, 2021-07-14:
You could try reverting https://github.com/openwrt/luci/commit/f919635310286b3b905aedb716d707da1ea2c811 and see if it solves your issue. If it does, please provide details about your browser, OS etc.

turboproc, 2021-07-14:
@jow- please see my comment against commit you referred to. Solution has been included.

jow-, 2021-07-15:
@turboproc - indeed, thank you very much for the details. Fix pushed with 7d9ab8d.

commit 7d9ab8d4 (2021-07-15): luci-base: fix off-by-one in cbi_update_table()

Fixes: #5189
Fixes: f919635310 ("luci-base: batch DOM updates to prevent slowdown")
Ref: https://github.com/openwrt/luci/commit/f919635310286b3b905aedb716d707da1ea2c811#commitcomment-53495580
closed 2021-07-15 by jow-