openwrt/luci #5202: dispatcher: "No password set!" dialog no longer presents a direct link to admin/system/admin

pmelange, 2021-07-19:
In the past, when the header.htm file notices that a password has not been set, it would warn the user with the text 
```
No password set!
There is no password set on this router. Please configure a root password to protect the web interface.
```
and would provide a direct link to admin/system/admin with the text "Go to password configuration...".

This link is not longer being shown.  It does not matter if the user has clicked on the "login" button or not.  The last seen release where this worked was 19.07.2.

I assume that something has changed with the dispatcher as the code which creates the link is
```
<% if disp.lookup("admin/system/admin") then %>
	<div class="right"><a class="btn" href="<%=url("admin/system/admin")%>"><%:Go to password configuration...%></a></div>
<% end %>
```

richb-hanover, 2021-07-30:
I just filed a [report about this in the Forum](https://forum.openwrt.org/t/cant-log-in-after-factory-install-no-password-set/102633) and described the workaround (`ssh` in, change the password by hand)

**But it's even worse than it appears.** Even after I logged in via `ssh` to set a password, I could not log in via LuCI using Firefox. There was some kind of problem (cookies?) that continually brought me back to that screen, even after I entered the correct password.

The (second) workaround was to use a different browser to connect to the router, and enter the correct password.

pmelange, 2021-07-31:
@richb-hanover The issue you presented is different from the one described in the original post.  Please make a new issue.  The problem is with session cookies and not the link to admin/system/admin

adelton, 2025-01-11:
With OpenWrt 24.10.0-rc5 (r28304-6dacba30a7), the `Go to password configuration...` link is there, rendered as a button on the Bootstrap theme:
[image]

Does it mean this issue has been addressed?

systemcrash, 2025-01-11:
Yes

closed 2025-01-11 by systemcrash