openwrt/luci #5206: Flip-flop in Associated Stations between IPv4 name and useless IPv6 name

timkgh, 2021-07-22:
Associated Stations seems to now show both IPv4 and IPv6 for clients. It used to be the case that only IPv4 was shown with the correct host name based on a reverse DNS lookup.

I do not use IPv6 on my network and these IPv6 addresses are all link local that sometimes don't resolve to any name and other times they resolve to useless names such as `Android-3.local` when the IPv4 resolves to a useful name, e.g. `myphone`.

What's worse is that as the Associated Stations page auto-refreshes, sometimes you see the IPv4 name other times the useless IPv6 name, it just keeps flip-flopping between the two.

Can Luci just show the name corresponding to the IPv4 and ignore names from link local IPv6 (those that start with `fe80::`)?

Or is there something in OpenWRT that I can turn off to not discover these IPv6 addresses at all?

DittelHome, 2021-09-20:
Same Problem, different description.... #5368

jow-, 2021-09-22:
Should've been addressed with 93bd93e220714276eff95ea51793bb2f826328e8, 45a398e1ed86c9706d47ff46be73bdaa303ff661 and af4352c07ef33ebb06a0c7f684f3f753b9c23f4f

Will be cherry-picked to openwrt-21.02 once it received more testing

closed 2021-09-22 by jow-

timkgh, 2021-09-22:
Awesome, thanks! Look forward to the next 21.02 release.

DittelHome, 2021-09-26:
I thought it come up with the update function, isnt it ?
Or do we have to wait until 21.02.1 ?