openwrt/luci #5212: luci-app-attendedsysupgrade: Error building the sysupgrade

richb-hanover, 2021-07-25:
Maintainer: @aparcar 
Environment: 21.02-rc1 on a Netgear WNDR3800. (Powered by LuCI openwrt-21.02 branch (git-21.106.55967-06dd6b5) / OpenWrt 21.02.0-rc1 r16046-59980f7aaf )

Description:

I was attempting to use Attended Sysupgrade from 21.02-rc1 on a Netgear WNDR3800. (Powered by LuCI openwrt-21.02 branch (git-21.106.55967-06dd6b5) / OpenWrt 21.02.0-rc1 r16046-59980f7aaf )

I chose System -> Attended Sysupgrade, and clicked **Search for sysupgrade**. That gave a modal window that stated that 21.02.0-r3 was available. Clicking **Request sysupgrade** gave me the following error message. 

What other information could I provide? Many thanks!

```
Error building the sysupgrade

asu.build.PackageSelectionError

Please report the error message and request
Request to server:

{
    "board_name": "netgear,wndr3800",
    "target": "ath79/generic",
    "version": "21.02.0-rc3",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip6tables",
        "iptables",
        "iw",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-cfg80211",
        "kmod-gpio-button-hotplug",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-leds-reset",
        "kmod-lib-crc-ccitt",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-owl-loader",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb-ohci",
        "kmod-usb2",
        "libblobmsg-json",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libustream-wolfssl",
        "libwolfssl",
        "libxtables",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "snmpd",
        "swconfig",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    ".anonymous": false,
    ".name": "request",
    ".type": "request",
    ".index": 0,
    "request_hash": "4e020882284d"
}

STDERR:

Downloading http://downloads.cdn.openwrt.org/releases/21.02.0-rc3/targets/ath79/generic/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/dl/openwrt_core
Downloading http://downloads.cdn.openwrt.org/releases/21.02.0-rc3/targets/ath79/generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.cdn.openwrt.org/releases/21.02.0-rc3/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/dl/openwrt_base
Downloading http://downloads.cdn.openwrt.org/releases/21.02.0-rc3/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.cdn.openwrt.org/releases/21.02.0-rc3/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/dl/openwrt_luci
Downloading http://downloads.cdn.openwrt.org/releases/21.02.0-rc3/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.cdn.openwrt.org/releases/21.02.0-rc3/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/dl/openwrt_packages
Downloading http://downloads.cdn.openwrt.org/releases/21.02.0-rc3/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.cdn.openwrt.org/releases/21.02.0-rc3/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/dl/openwrt_routing
Downloading http://downloads.cdn.openwrt.org/releases/21.02.0-rc3/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.cdn.openwrt.org/releases/21.02.0-rc3/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/dl/openwrt_telephony
Downloading http://downloads.cdn.openwrt.org/releases/21.02.0-rc3/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading https://feed.libremesh.org/master/Packages.gz
Updated list of available packages in /home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/dl/lime-packages
Downloading https://feed.libremesh.org/master/Packages.sig
Signature check passed.
Downloading https://feed.libremesh.org/profiles/Packages.gz
Updated list of available packages in /home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../home/aparcar/asu/cache/21.02.0-rc3/ath79/generic/dl/lime-profiles
Downloading https://feed.libremesh.org/profiles/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package libiwinfo.
make[2]: *** [Makefile:167: package_install] Error 255
make[1]: *** [Makefile:132: _call_manifest] Error 2
make: *** [Makefile:251: manifest] Error 2
```

aparcar, 2021-08-14:
@richb-hanover is this fixed? Please try rc4 and close this issue.

richb-hanover, 2021-08-14:
@aparcar  Yes, upgrading to rc4 worked a treat. I knew I wrote something about it... See: https://forum.openwrt.org/t/online-imagebuilder-and-upgrade-server/18110/295

Thanks!

closed 2021-08-14 by richb-hanover