openwrt/luci #5213: luci-app-firewall: Missing icmp-type dropdown options for bad-header and unknown-header-type

pavitter-bluestone, 2021-07-28:
I noticed that the default configuration of the firewall in openwrt includes the icmp-type bad-header and unknown-header-type for the 'Allow-ICMPv6-Input' and 'Allow-ICMPv6-Forward' ([link to the file in the current master branch](https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/network/config/firewall/files/firewall.config;h=5e22f984ce9f9e0ef3bd63f73816356c2dd1d8df;hb=ae1c5d0d6af60d0467899f5730a2f01aa72137f7)).

In the rules.js of luci-app-firewall both icmp-types are not a selectable option and when saving without any changes made, both icmp-types get removed. 

My question is, if this is intended behaviour? 

I tried search for both of those values and did not found anything. On the OpenWRT Wiki both types are absent too [OpenWRT Wiki Link to ICMP Types](https://openwrt.org/docs/guide-user/firewall/firewall_configuration#icmp_name_types).

systemcrash, 2021-08-31:
The wiki page you link to only seems to cover IPv4.

https://www.iana.org/assignments/icmpv6-parameters/icmpv6-parameters.xml

running this on my 21-rc4 build:
```
ip6tables -p ipv6-icmp -h
```

dumps
```
Valid ICMPv6 Types:
destination-unreachable
   no-route
   communication-prohibited
   beyond-scope
   address-unreachable
   port-unreachable
   failed-policy
   reject-route
packet-too-big
time-exceeded (ttl-exceeded)
   ttl-zero-during-transit
   ttl-zero-during-reassembly
parameter-problem
   bad-header
   unknown-header-type
   unknown-option
echo-request (ping)
echo-reply (pong)
router-solicitation
router-advertisement
neighbour-solicitation (neighbor-solicitation)
neighbour-advertisement (neighbor-advertisement)
redirect
```

Not sure what problems arise when one tries to assign ipv6 only types to IPv4 traffic when building rules. Might just be silently ignored, might not. Could be the reason why they were excluded, but that doesn't seem like the right approach.

systemcrash, 2021-08-31:
> Not sure what problems arise when one tries to assign ipv6 only types to IPv4 traffic when building rules. Might just be silently ignored, might not. Could be the reason why they were excluded, but that doesn't seem like the right approach.

Yeah, seems this is an omission. There are other ipv6 only types in there, so I created a PR which should resolve this in the master branch. Needs attention from some admins to land in 21.x

feckert, 2021-09-01:
Cherry-picked to openwrt-21.02 https://github.com/openwrt/luci/commit/3b3c2e5f9f82372df8ff01ac65668be47690dcd5

commit 3b3c2e5f (2021-09-01): luci-app-firewall: add ipv6 exclusive ICMP types

fixes #5213

(cherry picked from commit 19c66ef539da5136bcdb3978e5e8b87c87f8ae21)
Update commit description
closed 2021-09-01 by feckert