openwrt/luci #5219: Luci dependancy error libubus20191227

kevthehermit, 2021-07-31:
When trying to install Luci using image builder or manually I get a dependencies error for `libubus20191227`

Steps to reproduce. 

Use the image builder to build a generic armvirt-32 image

```
FROM openwrtorg/imagebuilder:armvirt-32-19.07.7
make image
```
Start the instance using qemu and try to install Luci or any of its mods

```
root@OpenWrt:/# opkg install luci
Installing luci (git-21.189.23240-7b931da-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/luci/luci_git-21.189.23240-7b931da-1_all.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci:
 * 	libubus20191227
 * opkg_install_cmd: Cannot install package luci.
```

Full Log with opkg update

```
root@OpenWrt:/# opkg update
Downloading http://downloads.openwrt.org/releases/19.07.7/targets/armvirt/32/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading http://downloads.openwrt.org/releases/19.07.7/targets/armvirt/32/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/freifunk/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_freifunk
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/freifunk/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.sig
Signature check passed.
root@OpenWrt:/# opkg install luci
Installing luci (git-21.189.23240-7b931da-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.7/packages/arm_cortex-a15_neon-vfpv4/luci/luci_git-21.189.23240-7b931da-1_all.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci:
 * 	libubus20191227
 * opkg_install_cmd: Cannot install package luci.
root@OpenWrt:/# 

```

kevthehermit, 2021-07-31:
```
root@OpenWrt:/# opkg install libubus
Package libubus20210603 (2021-07-01-38c7fdd8-1) installed in root is up to date.
root@OpenWrt:/# 

```

jow-, 2021-08-01:
Packaging error in OpenWrt base, nothing we can solve here.

closed 2021-08-01 by jow-