openwrt/luci #5220: luci-app-mwan3: does not add a new item in any of the configuration tabs

tjgonbor, 2021-08-01:
When I click on the [ADD] button, the next window to enter the data for the new item does not appear.
Since I did a fresh reinstall, this error appeared. I configured everything via CLI by editing the /etc/config/mwan3 file.
The target platform I am using is "Stable Release builds x86-64" with the image file "openwrt-19.07.7-x86-64-combined-ext4.img".
I have tried from another PC and also with different browsers. Other LuCI options such as Networks and Firewall work fine.

Thanks in advance

feckert, 2021-08-02:
It works for me.
I don't think the uci syntax fits.
Could you please check the configuration `/etc/config/mwan3`.

tjgonbor, 2021-08-02:
Hi feckert
Thankful for your prompt reply.
I am just finishing testing in a virtual machine on virtualbox obtaining the same results: the [ADD] button does not work in any of the tabs of the LuCI option http://<Openwrt host name>/cgi-bin/luci/admin/network/mwan

I will continue working on it by editing via text editor in ssh session of the configuration file /etc/config/mwan3

Thanks a lot.