openwrt/luci #5222: luci-app-attendedsysupgrade: Searching for an available sysupgrade hangs indefinitely if API is not available

jamesmacwhite, 2021-08-03:
I'm aware there's been some issues with http://chef.libremesh.org/ recently, but it seems when the API endpoints i.e. /api/latest or /api/build aren't available it is not handled gracefully in LuCI and will just sit on this screen (in the case of searching for a build).

[image]

A JS stacktrace will eventually be thrown in the background however the error has not been gracefully handled.

VM457:1 GET https://chef.libremesh.org/api/latest net::ERR_CONNECTION_CLOSED
(anonymous) @ VM457:1
check_sysupgrade @ overview.js?v=git-21.048.62294-9df7ea4:19
click @ overview.js?v=git-21.048.62294-9df7ea4:29
attendedsysupgrade:1 Uncaught (in promise) TypeError: Failed to fetch
Promise.then (async)
check_sysupgrade @ overview.js?v=git-21.048.62294-9df7ea4:19
click @ overview.js?v=git-21.048.62294-9df7ea4:29
```

Potentially it would be good from a user perspective to set a max timeout on the request to the chef endpoint and then display an error.

aparcar, 2021-08-09:
Sounds good! I'll try to figure out how to do it.

Server is up again, I've just tested it. However please consider using the new server at https://asu.aparcar.org

jamesmacwhite, 2021-08-09:
Thank you @aparcar!

I did notice the default server change in master a few days ago. If I change my configuration to this it would appear there may be a CORS issue on the API request however.

```
Access to fetch at 'https://asu.aparcar.org/api/latest' from origin 'https://linksys-wrt3200acm.internal.jmwhite.co.uk' has been blocked by CORS policy: The 'Access-Control-Allow-Origin' header contains the invalid value 'asu.aparcar.org'. Have the server send the header with a valid value, or, if an opaque response serves your needs, set the request's mode to 'no-cors' to fetch the resource with CORS disabled.
```

The chef.libremesh.org server works fine.

aparcar, 2021-08-09:
I created an PR to fix this, please see #5240

The CORS issue should be fixed. For the sake of (in)convenience I tried this shiny Caddyserver, CORS should be fixed (again).

jamesmacwhite, 2021-08-09:
Nice, that PR should allow for more graceful error handling going forward if the API endpoints don't respond in a timely fashion.

Can confirm, I can now see the `Access-Control-Allow-Origin` header and it's working when setting the server to asu.aparcar.org!

Caddy is a good choice, lightweight, fast and scalable!

PR #5240 (2021-08-09): luci-app-attendedsysupgrade: add timeout and catch
If the upgrade server API does not respond, show an error message.
Fix #5222

While at it, minimal code linting
commit ff24b78c (2021-08-09): luci-app-attendedsysupgrade: add timeout and catch

If the upgrade server API does not respond, show an error message.
Fix #5222

While at it, minimal code linting
commit 624578d3 (2021-08-23): luci-app-attendedsysupgrade: add timeout and catch

If the upgrade server API does not respond, show an error message.
Fix #5222

While at it, minimal code linting

(cherry picked from commit ff24b78c807a52a0f9afb62ba0aba1085d9ad045)
closed 2021-08-09 by aparcar