openwrt/luci #5232: Timezone cannot find China, suggest change "Shanghai" to "China"

RadioOperator, 2021-08-07:
Hi, refer to attached
[image]

stangri, 2021-08-07:
This comes from [tz database](https://en.wikipedia.org/wiki/List_of_tz_database_time_zones), you should bring it up with [ICANN](https://en.wikipedia.org/wiki/Tz_database) if you want this changed.

closed 2021-08-07 by stangri

RadioOperator, 2021-08-07:
@stangri Hi, thanks, I hope you/your openwrt could change this in further new release.

hnyman, 2021-08-08:
Like stangri said, the timezone names come from the globally standardised database, mandated by IANA / ICANN.

https://www.iana.org/time-zones

https://data.iana.org/time-zones/tz-link.html

> Timezones are typically identified by continent or ocean and then by the name of the largest city within the region containing the clocks. For example, America/New_York represents most of the US eastern time zone; America/Phoenix represents most of Arizona, which uses mountain time without daylight saving time (DST);

The standard is to use city names, so there will not be "Asia/China"

RadioOperator, 2021-08-08:
Copy that, thanks.