openwrt/luci #5234: LuCi remove required user option from wireless config

erdoukki, 2021-08-07:
The legacy rates options is still necessary and required for mwifiex and 8997.
This is the WiFi included in MVEBU EspressoBin Ultra...
Without the legacy rates, the Access Point do not work at all...
The option can be add by hand, or with uci command line, **but** any modification with LuCi will removed this option...
This is a really annoying !
How can we fix this ?

erdoukki, 2021-08-07:
May be this is this commit which is in fault : https://github.com/openwrt/luci/commit/f3417f287179d2918f90be1fcc8f7997dd7764d8

Here is my working (test) wireless config : 
```
root@ULTRA-5G:~# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'soc/d0070000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option htmode 'HT20'
	option hwmode '11g'
	option channel 'auto'
	option log_level '1'
	option cell_density '0'
	option country 'FR'
	option legacy_rates '1'
```
legacy_rates is not shown in luci and is removed if I do any modification and save the wireless config with luci

erdoukki, 2021-08-07:
```
root@ULTRA-5G:~# iwinfo wlan0 info
wlan0     ESSID: "TRIBU"
          Access Point: F0:AD:4E:10:0E:C3
          Mode: Master  Channel: 1 (2.412 GHz)
          Center Channel 1: 1 2: unknown
          Tx-Power: 15 dBm  Link Quality: 49/70
          Signal: -61 dBm  Noise: -91 dBm
          Bit Rate: 13.7 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 1B4B:2B42 0000:0000 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
```

hnyman, 2021-08-08:
@jow- 

> May be this is this commit which is in fault : f3417f2

"luci-mod-network: restrict legacy_rates option to hwmode 11g"

erdoukki might be right, and the commit f3417f2 combined to the later change of having "band" instead of "hwmode"  causes now the legacy_rates option to get disabled as the dependency does not get fulfilled.

https://github.com/openwrt/luci/blob/e5626ece12236f6be9dbb6da6eb90fcbb469a1f0/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js#L945

closed 2021-08-09 by jow-

erdoukki, 2021-08-10:
I confirm that the commit https://github.com/openwrt/luci/commit/119701cce302b6475e2fe900d2fff6c5dad16416 fix this !
The option is back again...
thanks @jow-

commit fd2e755b (2021-08-09): luci-mod-network: fix legacy_rates option dependency

The recently introduced band option support changed the internal frequency
selector choice values, breaking the legacy rates option dependency.

Fixes: #5234
Fixes: 0829d31290 ("luci-mod-network: support newer wireless.wifi-device.band option")
commit 119701cc (2021-08-09): luci-mod-network: fix legacy_rates option dependency

The recently introduced band option support changed the internal frequency
selector choice values, breaking the legacy rates option dependency.

Fixes: #5234
Fixes: 0829d31290 ("luci-mod-network: support newer wireless.wifi-device.band option")

(cherry picked from commit fd2e755b6f30d80ffdf15536669fdb739212ac96)