openwrt/luci #5238: Path to autoupdate | Ability to use URL in LuCI for flashing

BryanQuigley, 2021-08-09:
I opened this on the [forum](https://forum.openwrt.org/t/path-to-auto-update/103154), but realized that the majority of things initially I want to do are in luci.

My thought is the place to start is with "Ability to use URL in LuCI for flashing".  Are there any concerns with adding that feature?  (or suggestions on a better place to start). Thanks!

jow-, 2021-08-09:
No concerns adding this feature, feel free to open a PR implementing it.

closed 2021-08-09 by jow-

BryanQuigley, 2021-08-13:
Quick question on implementation.. I could make a brand new section for it, like:
[image]

(This may initially have less options then uploading a file..)

or I could add URL support to the ui:uploadFile function?

Thanks!
Bryan