openwrt/luci #5281: LuCI removing 'device' option from ModemManager interfaces

nickberry17, 2021-08-17:
When making any change to a luci-proto-modemmanager interface the required `device` attribute is deleted, breaking the configuration of that interface.  Is this related to something LuCI is doing in regards to DSA?

feckert, 2021-08-17:
I have seen this too.
But this was fixed with commit https://github.com/openwrt/luci/commit/10f02472c5fdab72615a7d3695e8f354811cd661#diff-56df355dcbbd3eaeb1ed044f96b8e956fa4cb13b5202987f6622057823c08616 in the master branch?
Which branch do you use?

The best solution would be from my point of view:
Rename this kind of device option to **ctl_device**.
We already use this naming in the **wwan** proto handler
https://github.com/openwrt/openwrt/blob/master/package/network/utils/wwan/files/wwan.sh#L101

The following proto handler uses the device option naming too.
https://github.com/openwrt/luci/blob/master/protocols/luci-proto-qmi/htdocs/luci-static/resources/protocol/qmi.js#L58
https://github.com/openwrt/luci/blob/master/protocols/luci-proto-ncm/htdocs/luci-static/resources/protocol/ncm.js#L62

jow-, 2021-08-19:
This has already been fixed by f3f74bd0fe66b94a99a8d944b63dcd6bdd1b93c6,  10f02472c5fdab72615a7d3695e8f354811cd661 and 96ee6dc8d6c9796ab67de6f313a068a4bd3bb20f

closed 2021-08-19 by jow-