openwrt/luci #5285: luci-app-bcp38 fails CAIDA bcp38 tests

tiagofreire-pt, 2021-08-17:
I did some tests using CAIDA software at different moments of luci-app-bcp38:

Before installing it: https://spoofer.caida.org/report.php?sessionid=1198440
After: https://spoofer.caida.org/report.php?sessionid=1198445

The configs:

[image]

I can't see any difference between both tests.

I'm using `OpenWrt 19.07.8 r11364-ef56c85848 / LuCI openwrt-19.07 branch git-21.189.23240-7b931da`

Am I doing anything wrong?

hnyman, 2021-08-17:
If I read that detailed path output right, you seem to be inside double NAT...
Apparently your ISP uses 10.27.x.x ???
That prevents BCP38 from working, as "your upstream" is still a NAT, so you can't really use BCP38.

```
Summary:
    Test run at: 2021-08-17 15:47:45 GMT
    Client Prefix (v4): 84.90.195.x/24
```
```
IP | Autonomous System
-- | --
Path 1 (to: 78.41.116.2)
192.168.20.1 | 0
10.27.0.1 | 0
84.91.1.221 | 13156
80.231.159.45 | 6453
```

tiagofreire-pt, 2021-08-17:
No, my ISP uses a public addressable IP address on the DHCP provided gateway:

# ip route
default via 84.90.192.1 dev eth1  src 84.90.195.xxx
84.90.192.0/22 dev eth1 scope link  src 84.90.195.xxx

tiagofreire-pt, 2021-08-17:
Disabling `Auto-detect upstream IP`: https://spoofer.caida.org/report.php?sessionkey=zw1roa9ylpm70l

No changes either.

tiagofreire-pt, 2021-08-17:
For what is worth, another test with another portuguese ISP using the same "lan-style-first-hop":

```
Path 4 (to: 130.206.158.142)
192.168.10.1	0
10.240.192.1	0
* * *	0
195.8.30.241	8657
195.8.0.198	8657
195.8.0.198	8657
195.66.226.161	0
* * *	0
* * *	0
* * *	0
62.40.98.72	20965
83.97.88.130	21320
130.206.245.93	766
130.206.195.2	766
130.206.158.2	766
130.206.158.50	766
130.206.158.142	766 
```

Using `10.240.192.1` as first-hop (**but not as gateway!**), passed 100% at these tests. The router is not running Openwrt, instead Opnsense. So, It seems the issue is with Openwrt+luci-app-bcp38 and not with this type of ISP first hops.

In: https://spoofer.caida.org/report.php?sessionid=1198489

jow-, 2021-08-19:
Not related to LuCI, please report at https://github.com/openwrt/packages/issues

closed 2021-08-19 by jow-

tiagofreire-pt, 2021-08-19:
Done, as suggested: https://github.com/openwrt/packages/issues/16402