openwrt/luci #5288: Rosy File Server 'File Server/Administration' button missing

arky, 2021-08-18:
The 'Rosy File Server/ Administration' button fails to appear on the file server views. It only appears in the admin views.

### Default File server view

[image]

### Admin view 

[image]

This might be a regression as this seems to be working during the original commit 
https://github.com/openwrt/luci/pull/2404#issue-241254643

arky, 2021-08-24:
@jow- @hnyman Any ideas how to resolve this regression?

MocconaCoffee, 2021-08-25:
Upvote, experiencing this issue presently

dibdot, 2021-08-30:
Maybe we should mark this unmaintained package as broken and remove the base package as well (move to packages-abandoned)?

closed 2021-11-20 by hnyman

arky, 2021-12-04:
@hnyman Could you please point me to where this package now resides in unmaintained packages? Thanks