openwrt/luci #5292: luci-app-wireguard: typographic error

castillofrancodamian, 2021-08-22:
It should be "hace 20s".

[image]

jow-, 2021-08-30:
The translation strings need to be changed in the source, using something like:
```javascript
		if (seconds < 60) {
			ago = _('%ds ago').format(seconds);
		} else if (seconds < 3600) {
			ago = _('%dm ago').format(seconds / 60);
		} else if (seconds < 86401) {
			ago = _('%dh ago').format(seconds / 3600);
```

This way translators can decide where to put the number.

An even better approach would be just doing this:

```javascript
ago = _('%t ago').format(seconds);
```

This will format the seconds value into something like `1h 12m 33s`, depending on the size of the value.

systemcrash, 2021-08-30:
> ```js
> ago = _('%t ago').format(seconds);
> ```

Does JS appropriately localize to the correct language specific abbreviations for H:M:S?

Also, should the .po files contain `%t`, according to your suggestion?

Then I can make a drive-by PR to fix

jow-, 2021-08-30:
> Does JS appropriately localize to the correct language specific abbreviations for H:M:S?

Not yet but I'll add that later on.

> Also, should the .po files contain %t, according to your suggestion?

Yes, they should contain

```gettext
msgid "%t ago"
msgstr "hace %t"
```

systemcrash, 2021-08-31:
> > Does JS appropriately localize to the correct language specific abbreviations for H:M:S?
> 
> Not yet but I'll add that later on.
> 

Would prefer if we don't push things back and depend on not-yet-done parts, and stick with what exists. 

Attempt: #5319

systemcrash, 2021-08-31:
> An even better approach would be just doing this:
> 
> ```js
> ago = _('%t ago').format(seconds);
> ```
> 
> This will format the seconds value into something like `1h 12m 33s ago`, depending on the size of the value.

The GUI currently does e.g.: Último Handshake: Tue, 31 Aug 2021 20:41:56 GMT (1m ago)

So this would present it in both absolute and relative.

castillofrancodamian, 2021-10-29:
> This issue can be closed.

Still not solved.

[image]

systemcrash, 2021-11-02:
Fixed in #5482 so it's in master.

closed 2021-11-03 by feckert