openwrt/luci #5317: What’s the logic the Theme applies on Openwrt MVC system?

netitgo, 2021-08-31:
I am desining my own theme of OPENWRT R21 version.
I rebuilded the `sysauth.htm` which is the loggin page and stored this files in my own project directory of which the path is: `/view/themes/myTheme/sysauth.htm`.
the system's original sysauth.htm was stored in the path `/view/sysauth.htm`.
When I was doing the work locally, it worked fine that the system will render my own sysauth.htm file in priority rather than his own original file, but after I compiled my theme into IPK and installed on the other OPENWRT system with the same R21 version, the system prefered to use its own sysauth.htm template file rather than mine, and the my layout was broken saddly.
Thus, I was wondering are there any logics that the MVC system applies to? and how could I get my own sysauth.htm work as default or in priority?
Thx!

jow-, 2021-08-31:
Did you try to clear all caches? Doing `rm -r /tmp/luci-*`, restarting uhttpd etc.?

netitgo, 2021-08-31:
yes, I have tried to clean the cache and restart the uhttpd, and even restart the OP system...
it's really out of my wits..

jow-, 2021-08-31:
Well, from that description I can only conclude that either the .ipk is wrongly packaged (sysauth.htm ended up in wrong path?) or that maybe some code minification process broke the template in a way that rendered it broken (if LuCI fails to compile the theme template it will silently fall back to the next candiate, the default one).

Carefully compare the sysauth.htm on your target device after ipk installation with the original one, see if you can spot any differences.

closed 2021-12-03 by jow-