openwrt/luci #5322: Architecture displayed on Status page for Raspberry Pi 3B is shown as ?

srezone, 2021-09-01:
Architecture displayed on Status page for Raspberry Pi 3B is shown as ?
And, the incorrect output should look more like this:
```
root@OpenWrt:~# ubus call system board
{
	"kernel": "4.14.241",
	"hostname": "OpenWrt",
	"model": "Raspberry Pi 3 Model B Rev 1.2",
	"board_name": "raspberrypi,3-model-b",
	"release": {
		"distribution": "OpenWrt",
		"version": "19.07.8",
		"revision": "r11364-ef56c85848",
		"target": "brcm2708/bcm2710",
		"description": "OpenWrt 19.07.8 r11364-ef56c85848"
	}
}
root@OpenWrt:~# 
```
The system web displays information like this：
```
Hostname: OpenWrt
Model: Raspberry Pi 3 Model B Rev 1.2
Architecture: ?
Firmware Version: OpenWrt 19.07.8 r11364-ef56c85848 / LuCI openwrt-19.07 branch git-21.238.28320-53f59d3
Kernel Version: 4.14.241
```

jow-, 2021-09-01:
Not a LuCI issue, the relevant information is not exposed by OpenWrt

closed 2021-09-01 by jow-

srezone, 2021-09-03:
> @srezone The json output of `ubus call system board` does not have the `system` value set.
> Pleas have a look at the procd source how the output gets build
> https://git.openwrt.org/?p=project/procd.git;a=blob;f=system.c;hb=773e8da41532b23888511cd864fbd251ecadc577#l138
> Maybe the output of `cat /proc/cpuinfo` does not match how procd need this.

Thanks feckert : )