openwrt/luci #5327: Luci/docker is broken

terrytw, 2021-09-03:
There is detailed discussion here:
https://forum.openwrt.org/t/snapshot-cgroup-options-break-docker/102531/4

I have run into the same problem. I assume that this is broken for everyone.

wulfy23, 2021-10-12:
@feckert

as the main kernel does not enable `# CONFIG_BLK_CGROUP_IOCOST` afaik the defaulting of blkio to 500 in dockerman seems to break stuff...

bobafetthotmail, 2021-11-11:
hm, in the above discussion (here translated to english https://github-com.translate.goog/lisaac/luci-app-dockerman/issues/131?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=de&_x_tr_pto=nui ) the upstream dev said that this was fixed with this commit in his repo https://github.com/lisaac/luci-app-dockerman/commit/75813385deb83113e9874ccb538f91db10c81046#diff-117047b2dc6450eae4622e1aadc0de43e3a5660b5be5e66b74ad122e15d08abc
that does not seem to be added to OpenWrt's luci-docker.
If someone wants to copy that over, it should be good enough for a fix.

It will allow the user to actually leave the field empty, and the container will no more require that kernel option.

acooler15, 2021-11-16:
There are too many differences between [openwrt/luci/applications/luci-app-dockerman](https://github.com/openwrt/luci/tree/master/applications/luci-app-dockerman) and [lisaac/luci-app-dockerman/applications/luci-app-dockerman](https://github.com/lisaac/luci-app-dockerman/tree/master/applications/luci-app-dockerman) that it's hard to synchronize now.

feckert, 2021-11-16:
I have already suggested that we have only one code base. https://github.com/lisaac/luci-app-dockerman/issues/93

bobafetthotmail, 2021-11-16:
> There are too many differences between [openwrt/luci/applications/luci-app-dockerman](https://github.com/openwrt/luci/tree/master/applications/luci-app-dockerman) and [...]

syncronizing the whole application is not required to fix this issue.

Just edit this file https://github.com/openwrt/luci/blob/master/applications/luci-app-dockerman/luasrc/model/cbi/dockerman/newcontainer.lua#L723

to reflect the same changes the developer did in the small commit he did to fix in his tree https://github.com/lisaac/luci-app-dockerman/commit/75813385deb83113e9874ccb538f91db10c81046#diff-117047b2dc6450eae4622e1aadc0de43e3a5660b5be5e66b74ad122e15d08abcR484

agrady42, 2022-01-14:
I'm having this issue as well. While waiting for a release that fixes this, is there something I can do in the UI? It looks like I have to enter a `Block IO Weight`, so it will never be `nil` (which is what I think I want), but I'm not at all sure if I'm thinking in the right direction.

terrytw, 2022-01-15:
> I'm having this issue as well. While waiting for a release that fixes this, is there something I can do in the UI? It looks like I have to enter a `Block IO Weight`, so it will never be `nil` (which is what I think I want), but I'm not at all sure if I'm thinking in the right direction.
> 
> Edit: or config files. I'm relatively new to Open WRT and Luci and am still learning my way around.

No, there is nothing you can do except waiting for a fix. Unfortunately nobody is working on it.

agrady42, 2022-01-15:
Thanks, at least I'm not missing anything. I may just manage docker from the command line for now, I need more practice there anyway.

everyplace, 2022-04-02:
I just wanted to present a workaround to this. I experienced this problem through the UI as well, but if I initially launched a container via the CLI while ssh'd into the router, then it worked. Subsequently, you can manage the container via the UI as one normally would.

myrlyn, 2022-10-17:
At least as late as 22.03.1, this appears to be a single-line change  in /usr/lib/lua/luci/model/cbi/dockerman/newcontainer.lua (which corresponds to /applications/luci-app-dockerman/luasrc/model/cbi/dockerman/newcontainer.lua in the source tree), change line 643 
from 
o.default = default_config.blkio_weight or 500
to
o.default = default_config.blkio_weight or nil
This changes the default value to nil for new containers if you leave the textbox empty.  
I have opened a pull request with this change after testing on my own machine.

terrytw, 2022-10-18:
I don't know how Openwrt handles it, but it would be really helpful if it can be backported to 21.02.

commit 848f8ffc (2022-10-18): luci-app-dockerman: change default to nil if data.blkio_weight is not defined

Fixes: #5327
closed 2022-10-18 by jow-