openwrt/luci #5333: luci-app-mwan3 rule  Can't support multiple ipv4 separated by a single line with a comma

openips, 2021-09-04:
with commit 9c4bc0472f92947fa19b3231f12e88ba0518bd03
luci-app-mwan3 rule can not support multiple ipv4 separated by a single line with a comma in source ip like A,B,C,D ABCD all are a ipv4 address
fix the bug thanks 

[image]

dibdot, 2021-09-12:
Doesn't look like a bug. The mentioned input field only accepts a single IP-Address (or a range in CIDR notation).

@feckert

openips, 2021-09-12:
mwan3 config file can  support multiple ipv4 separated by a single line with a comma in source ip

feckert, 2021-09-13:
@dibdot I have changed the lua app to javascript recently, during this process I must have made a mistake, so that the entry of several IPs with a comma no longer works.
This is because I have now set the type to IPv4.

@openips From my point of view, this should be a list element anyway, then I can add several again.
For this to work, I have to adapt the mwan3 and create an upgrade script that changes the option from comma separated to a list option.

From my point of view, the comma is a mistake, it is simply due to history that this was allowed until now.