openwrt/luci #5339: add target name to status overview

nerdoc, 2021-09-06:
Every time I update my Openwrt routers (3) to another newer version, I have to lookup the target again and search what it is.
I do the update per Luci, and it shows me a nice overview page of the current status
[image]

But on the openwrt homepage/[download page](https://downloads.openwrt.org/releases/19.07.8/targets/), there is no info about that model. sure, as many models use the same target.

But, couldn't you easily add the target name to that site, like so:
[image]

Would be a lot easier to find it. Ideally you could add a link to the upgrade page `/cgi-bin/luci/admin/system/flash` too - or even an automatic upgrade....
But for the start, the target name would be helpful.

nerdoc, 2021-09-06:
the target is not enough... In my case there are 2 different versions for the same target.
* Linksys WRT1900AC (Mamba)
* Linksys WRT1900ACv2 (Cobra)

It would be helpful to place these on the status page.

systemcrash, 2021-09-06:
Agreed. https://github.com/openwrt/luci/pull/5340

systemcrash, 2021-09-06:
ssh to your device and run `ubus call system board` - what does it return?

nerdoc, 2021-09-06:
```
{
        "kernel": "4.14.241",
        "hostname": "Linksys",
        "system": "ARMv7 Processor rev 2 (v7l)",
        "model": "Linksys WRT1900AC",
        "board_name": "linksys,mamba",
        "release": {
                "distribution": "OpenWrt",
                "version": "19.07.8",
                "revision": "r11364-ef56c85848",
                "target": "mvebu/cortexa9",
                "description": "OpenWrt 19.07.8 r11364-ef56c85848"
        }
}
```

aparcar, 2021-09-07:
Have a look at `luci-app-attendedsysupgrade`. The app should figure out the stuff on it's own and even installs all package modifications.

systemcrash, 2021-09-07:
Sweet. But... still nice to have. I sometimes need to poke around to figure out which MIPS type the CPU is. See linked PR above.

nerdoc, 2021-09-08:
@aparcar `luci-app-attendedsysupgrade` has no UI, as I can see... So it does this all on its own? (ther is [no documentation](https://github.com/openwrt/luci/tree/openwrt-19.07/applications/luci-app-attendedsysupgrade) there neither). This would be a solution for me as well - but adding the hints would help in any case.