openwrt/luci #5342: Required dependency package kmod-wireguard is not available in any repository

Stenfrank, 2021-09-07:
When updating TP-Link Archer C7 v5 to version 21.02.0 Wireguard stopped working due to package "kmod-wireguard"

[image]

alexforencich, 2021-09-08:
Just ran in to this issue myself.  Definitely a bug in luci, but if you click "install" it should actually install the package, despite the luci GUI indicating that a piece is missing.

jow-, 2021-09-08:
The actual bug was within the `cgi-io` helper which truncated the package lists downloaded by LuCI. Once the package has been rebuilt, you can solve the issue using `opkg update; opkg upgrade cgi-io`.

The next maintenance release will contain the fixed version of this package.

closed 2021-09-08 by jow-

Stenfrank, 2021-09-10:
Thank you very much, manage the update as you indicated and it worked, the only thing I see is that the Status Wireguard menu no longer appears

[image]

reporrted, 2021-09-13:
> Thank you very much, manage the update as you indicated and it worked, the only thing I see is that the Status Wireguard menu no longer appears
> 
> [image]

Install luci-app-wireguard

Stenfrank, 2021-09-13:
> > Muchas gracias, gestiona la actualización como indicaste y funcionó, lo único que veo es que ya no aparece el menú Status Wireguard
> > [image]
> 
> Instalar luci-app-wireguard

Thanks