openwrt/luci #5343: auto remove all available mac-address in MAC-FILTER after perform add to blacklist on Luci - Status - Overview

hongdat1106, 2021-09-07:
Please correct me if I mean this is a bug when I perform "add to blacklist" in Luci-Status-Overview then UCI will remove all old MAC-Address (all available mac-address) automatically.

`# /etc/config/wireless
uci del wireless.default_radio0.maclist
uci add_list wireless.default_radio0.maclist='00:00:00:00:00:00'`

images:
https://www.dropbox.com/sh/k1ww5uzvxtt49l2/AADpC8i5KzY-b2gs1U2h_lHVa?dl=0

jow-, 2021-09-12:
@Ansuel - afair you implemented this feature in ef94c52680ab2743627fb821e16b0bb202f7b35c, please check.

Ansuel, 2021-09-12:
Tell me if I'm right. When you click add to blacklist, all the other old entry are removed?

hongdat1106, 2021-09-13:
> @Ansuel - afair you implemented this feature in [ef94c52](https://github.com/openwrt/luci/commit/ef94c52680ab2743627fb821e16b0bb202f7b35c), please check.

Yes, I built firmware for myself on Openwrt github - master

hongdat1106, 2021-09-13:
> Tell me if I'm right. When you click add to blacklist, all the other old entry are removed?

Yes, correct. All old entries will be deleted after submit.

commit 085c4a08 (2021-09-22): luci-mod-status: fix wrong logic for maclist adding feature

Currently any new entry added to the black/whitelist deletes the
others. Fix this and improve the rule apply by firstly disconnect
the wifi client.

Fixes: #5343
closed 2021-09-22 by jow-