openwrt/luci #5344: Move luci-app-olsr2 from the Freifunk-Spalter/packages repo to the openwrt/luci repo

pmelange, 2021-09-07:
Currently there is an implementation of luci-app-olsr2 in https://github.com/Freifunk-Spalter/packages/tree/master/luci/luci-app-olsr2

Since this is not a freifunk specific app, it would make more sense for the app to be located in this repo

@XDjackieXD seems to be the most recent responsible person for luci-app-olsr2. https://github.com/Freifunk-Spalter/packages/issues/12#issuecomment-914262872

pmelange, 2021-09-07:
The code is also to be found here https://git.chaosfield.at/jakob/luci-olsrd2

hnyman, 2021-09-07:
We already have luci-app-olsr, so shouldn't it be updated to the new version?

Additionally, the new apps should preferably be done with javascript instead of lua.

pmelange, 2021-09-07:
Actually, olsr and olsr2 are two completely different and incompatible daemons. 

I totally agree with a rewrite in js.  I was hoping that @XDjackieXD was motivated to do it.

XDjackieXD, 2021-09-07:
I'm currently lacking time to work on another software project.
As for why I wrote it in lua this basically boils down to having the olsr1 app as an example and me being unable to find documentation on how to write luci apps with the "new" system