openwrt/luci #5347: Integrate luci-ssl and luci-app-ttyd

alexforencich, 2021-09-08:
It would be great if the same SSL certificates could be used for both luci-ssl and luci-app-ttyd.  I tried pointing luci-app-ttyd at the /etc/uhttpd.crt and /etc/uhttpd.key files, but unfortunately this does not work as it seems uhttpd.key is in some binary format that ttyd does not understand and hence ttyd does not start.  Not sure what the best solution to this would be - perhaps change the format for the automatically-generated uhttpd.key to something that both uhttpd and ttyd can understand, or alternatively to patch ttyd to understand whatever format is used for uhttpd.key.

alexforencich, 2021-09-09:
See also #5048

professor-jonny, 2021-11-20:
> As a side-note, installing luci-app-ttyd alongside luci-ssl without enabling SSL in luci-app-ttyd is rather useless as modern browsers tend to block the non-secure iframe that luci-app-ttyd uses to connect to ttyd when the page is served over HTTPS. So having a nice way to make this work out-of-the-box would be quite convenient.

Did you manage to generate compatable certs for ttyd and get it to work?
If so could you point me to this information ?

ysc3839, 2021-12-13:
I think the final solution is to use some http reverse proxy server(nginx) to allow apps to share a single SSL port(443). However it may not easy to generate config file automatically. Still requires manual configure.
I think enabling SSL is an advanced option because you have to use a domain name to get a CA signed certificate. Ignoring SSL error is unsafe and the error page will show again in some days after you ignore it. This is not user friendly and I think most users will not enable SSL.
So just leave this issue and let advanced users configure manually.

alexforencich, 2021-12-13:
Oh yeah, using a reverse proxy is a much better idea - no oddball open ports to deal with on the firewall end, no separate SSL configuration, and no issues with mixes SSL/non-SSL parts of the page.  It darn well better be possible to set that up automatically as part of the package so no manual intervention is required.

okibcn, 2022-03-29:
In any case, the compatibility between uhttpd and ttyd should be addressed. Creating all the SSL stuff is already complex. luci-uhttpd has managed to automate most of the tasks. @ysc3839, what would be the config files related to that alternative: I assume it involves installing more packages. At least `nginx` and `nginx-ssl`, isn't it?

alexforencich, 2022-03-29:
The problem is that the certificates generated for uhttpd are not in a format that ttyd can understand.  I already tried pointing ttyd at the certificates generated for uhttpd and ttyd did not start.  Unless that has been fixed, by either generating the certs with a different format (in which case they may need to be regenerated somehow) or by patching ttyd to support the current format.  Not sure if anything has changed since I originally opened this issue though, maybe you can do that now.

derzahla, 2024-01-21:
FYI: got this working by converting the uhttp cert and key to pem format.  for example:
'''
openssl x509 -inform der -in /etc/uhttpd.crt -out /etc/ttyd.crt
openssl ec -inform der -in /etc/uhttpd.key  -out ttyd.key
'''

Then set ttyd to use those instead of the uhttpd format

trigger2k20, 2024-04-05:
Unfortunatly with Openwrt version 23 the first converting command ist not working (anymore) _openssl x509 -inform der -in /etc/uhttpd.crt -out /etc/ttyd.crt_
Error: 281473146768480:error:0D0680A8:asn1 encoding routines:asn1_check_tlen:wrong tag:crypto/asn1/tasn_dec.c:1149:
281473146768480:error:0D07803A:asn1 encoding routines:asn1_item_embed_d2i:nested asn1 error:crypto/asn1/tasn_dec.c:309:Type=X509

Any possible ?

feckert, 2024-04-08:
This change is in my patchset for a long time. This could be helpfull. I could open a pullrequest to openwrt master to update the init script of uhttpd.
```
From: Florian Eckert <fe@dev.tdt.de>
Date: Tue, 18 Aug 2020 14:49:26 +0200
Subject: [PATCH] uhttpd: add uhttpd PEM generation

---
 package/network/services/uhttpd/files/uhttpd.init | 3 +++
 1 file changed, 3 insertions(+)

diff --git a/package/network/services/uhttpd/files/uhttpd.init b/package/network/services/uhttpd/files/uhttpd.init
index 31d7bcb18b5d522ee616480cfe9c5f7afbe4d8ad..1bedbc1da1789b54ef6ec49999a643bcba932e3e 100755
--- a/package/network/services/uhttpd/files/uhttpd.init
+++ b/package/network/services/uhttpd/files/uhttpd.init
@@ -61,6 +61,7 @@ generate_keys() {
 		sync
 		mv "${UHTTPD_KEY}.new" "${UHTTPD_KEY}"
 		mv "${UHTTPD_CERT}.new" "${UHTTPD_CERT}"
+		$OPENSSL_BIN x509 -inform der -in "${UHTTPD_CERT}" -out "${UHTTPD_PEM}"
 	}
 }
 
@@ -107,6 +108,7 @@ start_instance()
 {
 	UHTTPD_CERT=""
 	UHTTPD_KEY=""
+	UHTTPD_PEM=""
 
 	local cfg="$1"
 	local realm="$(uci_get system.@system[0].hostname)"
@@ -195,6 +197,7 @@ start_instance()
 	config_get https "$cfg" listen_https
 	config_get UHTTPD_KEY  "$cfg" key  /etc/uhttpd.key
 	config_get UHTTPD_CERT "$cfg" cert /etc/uhttpd.crt
+	config_get UHTTPD_PEM "$cfg" pem /etc/uhttpd.pem
 
 	[ -f /lib/libustream-ssl.so ] && [ -n "$https" ] && {
 		[ -s "$UHTTPD_CERT" -a -s "$UHTTPD_KEY" ] || {
-- 

```

trigger2k20, 2024-04-09:
Please go ahead :-)

trigger2k20, 2024-05-01:
Hi, any Updates ? @feckert

systemcrash, 2024-05-01:
> Unfortunatly with Openwrt version 23 the first converting command ist not working (anymore) _openssl x509 -inform der -in /etc/uhttpd.crt -out /etc/ttyd.crt_ 

@trigger2k20 
Here is your input format problem - while you assert an input form of `der`:
> Type=X509

@feckert does uhttpd know what to do with a PEM by itself in the init script already?

Neustradamus, 2024-08-16:
@alexforencich: Any progress on it?

phiberoptick, 2025-05-11:
Might want to try with the -outform flag:

`openssl x509 -inform der -in uhttpd.crt -outform pem -out ttyd.crt`

Or just convert the uhttpd cert and key:

`openssl x509 -inform der -in /etc/uhttpd.crt -outform pem -out /tmp/uhttpd.crt`
`openssl ec -inform der -in /etc/uhttpd.key -outform pem -out /tmp/uhttpd.key`

wryun, 2026-09-03:
https://github.com/openwrt/uhttpd/pull/41 might be helpful for this.