openwrt/luci #5357: MAC address is not set for secondary/new SSID causes radio to fail startup

alonbl, 2021-09-12:
Hi,

Just upgraded to:
```
OpenWrt 21.02.0 r16279-5cc0535800 / LuCI openwrt-21.02 branch git-21.231.26241-422c175
Linksys WRT3200ACM
```

I setup my guest SSID as 2nd SSID of the radio interface.
Once added, it causes the entire radio interface to fail startup.

I found that the macaddr statement at /etc/config/wireless is missing for the new entry.
I am not sure how it should be generated and if the luci is responsible for that... but I guess so.
Once I manually put a mac address everything is working correctly.

```patch
 config wifi-iface 'wifinet3'
         option device 'radio1'
         option mode 'ap'
         option ssid 'XXXX'
+        option macaddr 'e8:9f:80:1b:70:db'
         option isolate '1'
         option encryption 'psk2+ccmp'
         option key 'XXXX'
```

Everything is not actually everything... after restart the new interface is failing with `DEVICE_CLAIM_FAILED` error, manual restart to the interface fixes the issue, however, this is problematic as the system fails every power-down. But I guess this is not luci issue.

Regards,
Alon

alonbl, 2021-09-12:
In order to simulate the manual restart I can execute to workaround the issue of `DEVICE_CLAIM_FAILED` post restart:
```
wifi down radio1 && wifi up radio1
```

dibdot, 2021-09-12:
According to https://forum.openwrt.org/t/wrt3200acm-21-02-0-rc4-multiple-wifis-on-a-single-radio/103515 just remove these macaddr options from all configured SSIDs. Not a LuCI issue at all.

closed 2021-09-12 by dibdot