openwrt/luci #5361: /usr/lib/lua/luci/dispatcher.lua:427: /etc/config/luci seems to be corrupt, unable to find section 'main'

erdoukki, 2021-09-15:
I get errors in snapshots, 21.02.0-RC3 and now, still present on 21.02.0 : 
```
/usr/lib/lua/luci/dispatcher.lua:427: /etc/config/luci seems to be corrupt, unable to find section 'main'
```
This error appears on LuCI access on the Services/LXC page...
In logread I get some usefull informations : 
```
Wed Sep 15 09:36:55 2021 daemon.err uhttpd[1935]: luci: accepted login on /admin/services/lxc for root from 10.4.2.112
Wed Sep 15 09:36:56 2021 daemon.info procd: Instance rpcd::instance1 s in a crash loop 6 crashes, 104 seconds since last crash
```
How can I fix this or help to debug ?

Thanks in advance...

erdoukki, 2021-09-15:
The install is a fresh 21.02.0, without any old kept configs.

I have installed some packages, mostly LXC, and I can see there is a rpcd-mod-lxc necessary package for luci-app-lxc which have been installed also...

The problem only appears in luci-app-lxc !

erdoukki, 2021-09-15:
```
root@ULTRA-CLOUD:/srv/lxc# ubus -v list lxc 
'lxc' @df0b6bea
	"start":{"name":"String","config":"String"}
	"reboot":{"name":"String","config":"String"}
	"shutdown":{"name":"String","config":"String","timeout":"Integer"}
	"stop":{"name":"String","config":"String"}
	"freeze":{"name":"String","config":"String"}
	"unfreeze":{"name":"String","config":"String"}
	"info":{"name":"String","config":"String"}
	"rename":{"name":"String","config":"String","newname":"String"}
	"create":{"name":"String","config":"String","template":"String","flags":"Integer","args":"Array"}
	"destroy":{"name":"String","config":"String"}
	"list":{}
```

```
root@ULTRA-CLOUD:/srv/lxc# ubus call lxc list
Command failed: Request timed out

```

erdoukki, 2021-09-15:
```
root@ULTRA-CLOUD:/srv/lxc# lxc-ls -f
Failed to create lock for myMIAB
Failed to create lock for myMIB
Failed to create lock for myNextCloud
```

erdoukki, 2021-09-15:
before : 

```
root@ULTRA-CLOUD:/srv/lxc# lxc-checkconfig 
LXC version 4.0.5
--- Namespaces ---
Namespaces: enabled
Utsname namespace: enabled
Ipc namespace: enabled
Pid namespace: enabled
User namespace: enabled
Network namespace: enabled

--- Control groups ---
Cgroups: enabled

Cgroup v1 mount points: 

Cgroup v2 mount points: 
/sys/fs/cgroup

Cgroup v1 systemd controller: missing
Cgroup v1 freezer controller: missing
Cgroup namespace: required
Cgroup device: missing
Cgroup sched: enabled
Cgroup cpu account: enabled
Cgroup memory controller: enabled
Cgroup cpuset: enabled

--- Misc ---
Veth pair device: enabled, loaded
Macvlan: enabled, not loaded
Vlan: enabled, not loaded
Bridges: enabled, not loaded
Advanced netfilter: enabled, not loaded
CONFIG_NF_NAT_IPV4: missing
CONFIG_NF_NAT_IPV6: missing
CONFIG_IP_NF_TARGET_MASQUERADE: missing
CONFIG_IP6_NF_TARGET_MASQUERADE: missing
CONFIG_NETFILTER_XT_TARGET_CHECKSUM: enabled, not loaded
CONFIG_NETFILTER_XT_MATCH_COMMENT: enabled, loaded
FUSE (for use with lxcfs): enabled, not loaded

--- Checkpoint/Restore ---
checkpoint restore: missing
CONFIG_FHANDLE: enabled
CONFIG_EVENTFD: enabled
CONFIG_EPOLL: enabled
CONFIG_UNIX_DIAG: missing
CONFIG_INET_DIAG: missing
CONFIG_PACKET_DIAG: missing
CONFIG_NETLINK_DIAG: enabled
File capabilities: 

Note : Before booting a new kernel, you can check its configuration
usage : CONFIG=/path/to/config /usr/bin/lxc-checkconfig

```
```
root@ULTRA-CLOUD:/srv/lxc# service cgroupfs-mount enable
root@ULTRA-CLOUD:/srv/lxc# service cgroupfs-mount restart
```
after : 
```
root@ULTRA-CLOUD:/srv/lxc# lxc-checkconfig 
LXC version 4.0.5
--- Namespaces ---
Namespaces: enabled
Utsname namespace: enabled
Ipc namespace: enabled
Pid namespace: enabled
User namespace: enabled
Network namespace: enabled

--- Control groups ---
Cgroups: enabled

Cgroup v1 mount points: 

Cgroup v2 mount points: 
/sys/fs/cgroup

Cgroup v1 systemd controller: missing
Cgroup v1 freezer controller: missing
Cgroup namespace: required
Cgroup device: missing
Cgroup sched: enabled
Cgroup cpu account: enabled
Cgroup memory controller: enabled
Cgroup cpuset: enabled

--- Misc ---
Veth pair device: enabled, loaded
Macvlan: enabled, not loaded
Vlan: enabled, not loaded
Bridges: enabled, not loaded
Advanced netfilter: enabled, not loaded
CONFIG_NF_NAT_IPV4: missing
CONFIG_NF_NAT_IPV6: missing
CONFIG_IP_NF_TARGET_MASQUERADE: missing
CONFIG_IP6_NF_TARGET_MASQUERADE: missing
CONFIG_NETFILTER_XT_TARGET_CHECKSUM: enabled, not loaded
CONFIG_NETFILTER_XT_MATCH_COMMENT: enabled, loaded
FUSE (for use with lxcfs): enabled, not loaded

--- Checkpoint/Restore ---
checkpoint restore: missing
CONFIG_FHANDLE: enabled
CONFIG_EVENTFD: enabled
CONFIG_EPOLL: enabled
CONFIG_UNIX_DIAG: missing
CONFIG_INET_DIAG: missing
CONFIG_PACKET_DIAG: missing
CONFIG_NETLINK_DIAG: enabled
File capabilities: 

Note : Before booting a new kernel, you can check its configuration
usage : CONFIG=/path/to/config /usr/bin/lxc-checkconfig

root@ULTRA-CLOUD:/srv/lxc# nano /etc/init.d/lxc-auto 
root@ULTRA-CLOUD:/srv/lxc# sevice lxc-auto enable
-ash: sevice: not found
root@ULTRA-CLOUD:/srv/lxc# service lxc-auto enable
root@ULTRA-CLOUD:/srv/lxc# service lxc-auto boot
root@ULTRA-CLOUD:/srv/lxc# lxc-checkconfig 
LXC version 4.0.5
--- Namespaces ---
Namespaces: enabled
Utsname namespace: enabled
Ipc namespace: enabled
Pid namespace: enabled
User namespace: enabled
Network namespace: enabled

--- Control groups ---
Cgroups: enabled

Cgroup v1 mount points: 
/sys/fs/cgroup/systemd

Cgroup v2 mount points: 
/sys/fs/cgroup

Cgroup v1 freezer controller: missing
Cgroup v1 clone_children flag: enabled
Cgroup device: missing
Cgroup sched: enabled
Cgroup cpu account: enabled
Cgroup memory controller: enabled
Cgroup cpuset: enabled

--- Misc ---
Veth pair device: enabled, loaded
Macvlan: enabled, not loaded
Vlan: enabled, not loaded
Bridges: enabled, not loaded
Advanced netfilter: enabled, not loaded
CONFIG_NF_NAT_IPV4: missing
CONFIG_NF_NAT_IPV6: missing
CONFIG_IP_NF_TARGET_MASQUERADE: missing
CONFIG_IP6_NF_TARGET_MASQUERADE: missing
CONFIG_NETFILTER_XT_TARGET_CHECKSUM: enabled, not loaded
CONFIG_NETFILTER_XT_MATCH_COMMENT: enabled, loaded
FUSE (for use with lxcfs): enabled, not loaded

--- Checkpoint/Restore ---
checkpoint restore: missing
CONFIG_FHANDLE: enabled
CONFIG_EVENTFD: enabled
CONFIG_EPOLL: enabled
CONFIG_UNIX_DIAG: missing
CONFIG_INET_DIAG: missing
CONFIG_PACKET_DIAG: missing
CONFIG_NETLINK_DIAG: enabled
File capabilities: 

Note : Before booting a new kernel, you can check its configuration
usage : CONFIG=/path/to/config /usr/bin/lxc-checkconfig

```
```
root@ULTRA-CLOUD:/srv/lxc# lxc-ls -f
NAME        STATE   AUTOSTART GROUPS IPV4 IPV6 UNPRIVILEGED 
myMIAB      STOPPED 0         -      -    -    false        
myMIB       STOPPED 0         -      -    -    false        
myNextCloud STOPPED 0         -      -    -    false        
root@ULTRA-CLOUD:/srv/lxc# ubus call lxc list
{
	"myMIAB": "STOPPED",
	"myMIB": "STOPPED",
	"myNextCloud": "STOPPED"
}
```

erdoukki, 2021-09-15:
Now; luci LXC access is OK !
So fixed...
I presume by : 
```
# service lxc-auto enable
# service lxc-auto boot
```

erdoukki, 2021-09-15:
Root cause : Cgroup namespace: required

erdoukki, 2021-09-15:
Have another problem on 21.02.0-RC3, on another installation... same issue, but not resolved (for now) : 
```
root@STARGATE:~# lxc-ls -f
Failed to load config for myBuster
Failed to load config for myMITM
Failed to load config for myMeshCentral
Failed to load config for myNC
Failed to load config for myNZB
Failed to load config for myNextCloud
Failed to load config for myOpenWisp
Failed to load config for myPICO
Failed to load config for myPICOSRV
Failed to load config for myPROXY
Failed to load config for myStretch
NAME       STATE   AUTOSTART GROUPS IPV4       IPV6 UNPRIVILEGED 
myCrowdSec STOPPED 0         -      -          -    false        
myDeeMix   RUNNING 1         -      10.4.2.119 -    false        
myLMS      RUNNING 1         -      10.4.2.173 -    false        
myMM2      STOPPED 1         -      -          -    false        
myREVERSE  RUNNING 1         -      10.4.2.16  -    false        
```
```
root@STARGATE:~# ubus call lxc list
Command failed: Request timed out

```

erdoukki, 2021-09-15:
```
root@STARGATE:~# lxc-ls -f -l=DEBUG
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 79 No such file or directory - Failed to open file "/usr/share/lxc/config/debian.common.conf"
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 121 Failed to parse config file "/srv/lxc/myBuster/config" at line "lxc.include = /usr/share/lxc/config/debian.common.conf"
Failed to load config for myBuster
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 79 No such file or directory - Failed to open file "/usr/share/lxc/config/debian.common.conf"
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 121 Failed to parse config file "/srv/lxc/myMITM/config" at line "lxc.include = /usr/share/lxc/config/debian.common.conf"
Failed to load config for myMITM
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 79 No such file or directory - Failed to open file "/usr/share/lxc/config/debian.common.conf"
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 121 Failed to parse config file "/srv/lxc/myMeshCentral/config" at line "lxc.include = /usr/share/lxc/config/debian.common.conf"
Failed to load config for myMeshCentral
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 79 No such file or directory - Failed to open file "/usr/share/lxc/config/debian.common.conf"
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 121 Failed to parse config file "/srv/lxc/myNC/config" at line "lxc.include = /usr/share/lxc/config/debian.common.conf"
Failed to load config for myNC
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 79 No such file or directory - Failed to open file "/usr/share/lxc/config/debian.common.conf"
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 121 Failed to parse config file "/srv/lxc/myNZB/config" at line "lxc.include = /usr/share/lxc/config/debian.common.conf"
Failed to load config for myNZB
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 79 No such file or directory - Failed to open file "/usr/share/lxc/config/debian.common.conf"
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 121 Failed to parse config file "/srv/lxc/myNextCloud/config" at line "lxc.include = /usr/share/lxc/config/debian.common.conf"
Failed to load config for myNextCloud
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 79 No such file or directory - Failed to open file "/usr/share/lxc/config/debian.common.conf"
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 121 Failed to parse config file "/srv/lxc/myOpenWisp/config" at line "lxc.include = /usr/share/lxc/config/debian.common.conf"
Failed to load config for myOpenWisp
lxc-ls: confile.c: parse_line: 2795 Unknown configuration key "lxc.kmsg"
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 121 Failed to parse config file "/srv/lxc/myPICO/config" at line "lxc.kmsg = 0"
Failed to load config for myPICO
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 79 No such file or directory - Failed to open file "/usr/share/lxc/config/debian.common.conf"
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 121 Failed to parse config file "/srv/lxc/myPICOSRV/config" at line "lxc.include = /usr/share/lxc/config/debian.common.conf"
Failed to load config for myPICOSRV
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 79 No such file or directory - Failed to open file "/usr/share/lxc/config/debian.common.conf"
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 121 Failed to parse config file "/srv/lxc/myPROXY/config" at line "lxc.include = /usr/share/lxc/config/debian.common.conf"
Failed to load config for myPROXY
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 79 No such file or directory - Failed to open file "/usr/share/lxc/config/debian.common.conf"
lxc-ls: parse.c: lxc_file_for_each_line_mmap: 121 Failed to parse config file "/srv/lxc/myStretch/config" at line "lxc.include = /usr/share/lxc/config/debian.common.conf"
Failed to load config for myStretch
NAME       STATE   AUTOSTART GROUPS IPV4       IPV6 UNPRIVILEGED 
myCrowdSec STOPPED 0         -      -          -    false        
myDeeMix   RUNNING 1         -      10.4.2.119 -    false        
myLMS      RUNNING 1         -      10.4.2.173 -    false        
myMM2      STOPPED 1         -      -          -    false        
myREVERSE  RUNNING 1         -      10.4.2.16  -    false        

```

erdoukki, 2021-09-15:
What I have fixed from my errors in different configs : 

```
# Distribution configuration
##lxc.include = /usr/share/lxc/config/debian.common.conf
lxc.include = /usr/share/lxc/config/common.conf
lxc.arch = linux64
```

```
##lxc.network.hwaddr = 00:FF:AA:00:00:06
lxc.net.0.hwaddr = 00:FF:AA:00:00:06
```

```
##lxc.kmsg = 0
```

after : 

```
root@STARGATE:~# lxc-ls -f -l=DEBUG
NAME          STATE   AUTOSTART GROUPS IPV4       IPV6 UNPRIVILEGED 
myBuster      STOPPED 0         -      -          -    false        
myCrowdSec    STOPPED 0         -      -          -    false        
myDeeMix      RUNNING 1         -      10.4.2.119 -    false        
myLMS         RUNNING 1         -      10.4.2.173 -    false        
myMITM        STOPPED 1         -      -          -    false        
myMM2         STOPPED 1         -      -          -    false        
myMeshCentral STOPPED 0         -      -          -    false        
myNC          STOPPED 0         -      -          -    false        
myNZB         STOPPED 0         -      -          -    false        
myNextCloud   STOPPED 0         -      -          -    false        
myOpenWisp    STOPPED 1         -      -          -    false        
myPICO        STOPPED 1         -      -          -    false        
myPICOSRV     STOPPED 0         -      -          -    false        
myPROXY       STOPPED 1         -      -          -    false        
myREVERSE     RUNNING 1         -      10.4.2.16  -    false        
myStretch     STOPPED 0         -      -          -    false        
```

```
root@STARGATE:~# ubus call lxc list
{
	"myBuster": "STOPPED",
	"myCrowdSec": "STOPPED",
	"myDeeMix": "RUNNING",
	"myLMS": "RUNNING",
	"myMITM": "STOPPED",
	"myMM2": "STOPPED",
	"myMeshCentral": "STOPPED",
	"myNC": "STOPPED",
	"myNZB": "STOPPED",
	"myNextCloud": "STOPPED",
	"myOpenWisp": "STOPPED",
	"myPICO": "STOPPED",
	"myPICOSRV": "STOPPED",
	"myPROXY": "STOPPED",
	"myREVERSE": "RUNNING",
	"myStretch": "STOPPED"
}
```

Issue now fixed in LUCI-APP-LXC !

root cause : ubus / rpcd script of lxc with errors from lxc commands broke luci !?

erdoukki, 2021-09-15:
Problem may occurs with : 
```
# ubus call lxc list
Command failed: Request timed out
```

Possibly from rpcd-mod-lxc package...

erdoukki, 2021-09-15:
May be https://github.com/openwrt/packages/blob/master/utils/rpcd-mod-lxc/files/lxc.c must manage more efficiently timeout and errors... ?!

jow-, 2021-09-22:
Maybe, in any case it does not appear to be a LuCI issue. That lxc plugin fro mthe packages feed is crashing rpcd.

closed 2021-09-22 by jow-

erdoukki, 2021-09-22:
Sure, I agree with you @jow- 
It is a luci-app-lxc package error !
@ratkaj as the Maintainer, can you look at this issue, please ?

erdoukki, 2021-09-22:
@robimarko I read in https://github.com/openwrt/packages/issues/7694 that you also Maintainer of lxc ?

robimarko, 2021-09-22:
No, I just did that single update

erdoukki, 2021-09-22:
oups, okay, thanks...
And sorry for the ping !