openwrt/luci #5362: [RFC] recommended linter settings for JavaScript

aparcar, 2021-09-15:
Hi, I've tried to lint my LuCI app in the past but it failed since all tools somewhat messed up in awkward ways. So far I've tried `prettier`, `js-beautify` and `clang-format`. No really happy with any of them but now found this [online linter](https://beautifier.io/) with a bunch of options, does anyone have recommendations what to use?

feckert, 2021-09-23:
If I read this correctly, then this is [js-beautify](https://github.com/beautify-web/js-beautify).
I would prefer if we had a CI that started a linter with every pullrequest.
Of course, we first have to select a linter and also work out the configuration.

aparcar, 2021-09-23:
Ideally @jow- shares some wisdom here, however I think he doesn't use linters at this point but does things manually.

nickberry17, 2021-11-01:
Would be nice to standardise this so we can just set our editors up to format on save.

jow-, 2022-03-10:
I don't use any linter so can't recommend anything

closed 2022-03-10 by jow-