openwrt/luci #5373: luci-app-statistics: misconfiguration shipped pointing to non-existent directory

graysky2, 2021-09-18:
Maintainer: @jow- @feckert @hnyman 
Description:  A fresh install of luci-app-statistics expects `/etc/collectd/conf.d/` to be created on the file system but the package does not create this target.  You can see output in `logread` to this effect:
```
daemon.err collectd[23674]: configfile: stat (/etc/collectd/conf.d) failed: No such file or directory
```

Related but closed:
https://github.com/openwrt/packages/issues/12138
https://github.com/openwrt/luci/issues/439

feckert, 2021-09-21:
The application luci-app-statistics does **not** use the uci generation from the [collectd package](https://github.com/openwrt/packages/blob/master/utils/collectd/files/collectd.init).
The collectd configuration is generated by luci-app-statistics it self with the script [stat-genconfig](https://github.com/openwrt/luci/blob/master/applications/luci-app-statistics/root/usr/bin/stat-genconfi).
So I think we have to fix this there.

hnyman, 2021-09-21:
I beg to differ.

> A fresh install of luci-app-statistics expects /etc/collectd/conf.d/ to be created on the file system

Better explanation would be that collectd daemon itself expects that dir, but our collectd installation in Makefile or init script does not generate it.

Luci-statistics ships a config for collectd from here, but that does not have much role here.
The error is in the unnecessary warning as the collectd package itself does not check for the existence of the directory before looking into its contents.

Collectd init script (or the collectd package installation in Makefile) should probably generate that dir.

Probably here:
https://github.com/openwrt/packages/blob/master/utils/collectd/Makefile#L354-L366

Ps. that error seems to originate from collectd daemon itself, but at the first glance I do not see, from where exactly. (and the error has been there for a decade, I think.)

commit 2c27f5c9 (2021-10-05): luci-app-statistics: Remove erroneous include from default config

Remove the unnecessary Include line from the default config.
It causes error at collectd start, as the directory is missing.
  daemon.err collectd[9722]: configfile: stat (/etc/collectd/conf.d) failed: No such file or directory

Leave the line as a comment/example.

Fixes #5373
commit cb0574fc (2021-10-05): luci-app-statistics: Remove erroneous include from default config

Remove the unnecessary Include line from the default config.
It causes error at collectd start, as the directory is missing.
  daemon.err collectd[9722]: configfile: stat (/etc/collectd/conf.d) failed: No such file or directory

Leave the line as a comment/example.

Fixes #5373

(cherry picked from commit 2c27f5c9417ff5414a2730a9c310b68fdd5f585c)
closed 2021-10-05 by hnyman