openwrt/luci #5390: luci-attended-sysupgrade:

CAFxX, 2021-09-26:
I'm on a archer c7 v4 running `LuCI openwrt-19.07 branch (git-21.044.30835-34e0d65) / OpenWrt 19.07.7 r11306-c4a6851c72`, `attendedsysupgrade-common	0.2-1` and `luci-app-attendedsysupgrade	git-21.267.59215-1524972-1`; when I try to run attended sysupgrade to 19.07.8 from luci using the defaults, I get the following error:

```
Error building the sysupgrade

Unsupported package(s): libubus20191227
Please report the error message and request
Request to server:
{
    "board_name": "tplink,archer-c7-v4",
    "target": "ath79/generic",
    "version": "19.07.8",
    "packages": [
        "ath10k-firmware-qca988x-ct",
        "base-files",
        "busybox",
        "cgi-io",
        "collectd-mod-conntrack",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip6tables",
        "iptables",
        "iw",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-cfg80211",
        "kmod-gpio-button-hotplug",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-lib-crc-ccitt",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-core",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb-storage",
        "kmod-usb2",
        "libblobmsg-json",
        "libip4tc2",
        "libip6tc2",
        "libiwinfo-lua",
        "libiwinfo20181126",
        "libjson-c2",
        "libjson-script",
        "liblua5.1.5",
        "liblucihttp-lua",
        "liblucihttp0",
        "libnl-tiny",
        "libubox20191228",
        "libubus-lua",
        "libubus20191227",
        "libuci20130104",
        "libuclient20160123",
        "libustream-mbedtls20150806",
        "libxtables12",
        "logd",
        "lua",
        "luci",
        "luci-app-adblock",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-statistics",
        "luci-app-upnp",
        "luci-base",
        "luci-i18n-adblock-en",
        "luci-i18n-upnp-en",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "miniupnpd",
        "mtd",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "swconfig",
        "tcpdump-mini",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "uhttpd",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic"
    ],
    ".anonymous": false,
    ".name": "request",
    ".type": "request",
    ".index": 0
}
```

CAFxX, 2021-09-26:
Apparently in 19.07.08 libubus has been updated to libubus20210603

[image]

So if I enable advanced mode and manually drop libubus20191227 from the list, it runs but then runs into this error (that again I am reporting as instructed):

```
Error building the sysupgrade

Please report the error message and request
Request to server:
{
    "board_name": "tplink,archer-c7-v4",
    "target": "ath79/generic",
    "version": "19.07.8",
    "packages": [
        "ath10k-firmware-qca988x-ct",
        "base-files",
        "busybox",
        "cgi-io",
        "collectd-mod-conntrack",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip6tables",
        "iptables",
        "iw",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-cfg80211",
        "kmod-gpio-button-hotplug",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-lib-crc-ccitt",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-core",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb-storage",
        "kmod-usb2",
        "libblobmsg-json",
        "libip4tc2",
        "libip6tc2",
        "libiwinfo-lua",
        "libiwinfo20181126",
        "libjson-c2",
[... 73 lines cut ...]
Updated list of available packages in /home/debian/asu/worker2/cache/19.07.8/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/debian/asu/worker2/cache/19.07.8/ath79/generic/dl/openwrt_base
Downloading http://downloads.cdn.openwrt.org/releases/19.07.8/packages/mips_24kc/freifunk/Packages.gz
Updated list of available packages in /home/debian/asu/worker2/cache/19.07.8/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/debian/asu/worker2/cache/19.07.8/ath79/generic/dl/openwrt_freifunk
Downloading http://downloads.cdn.openwrt.org/releases/19.07.8/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /home/debian/asu/worker2/cache/19.07.8/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/debian/asu/worker2/cache/19.07.8/ath79/generic/dl/openwrt_luci
Downloading http://downloads.cdn.openwrt.org/releases/19.07.8/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /home/debian/asu/worker2/cache/19.07.8/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/debian/asu/worker2/cache/19.07.8/ath79/generic/dl/openwrt_packages
Downloading http://downloads.cdn.openwrt.org/releases/19.07.8/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /home/debian/asu/worker2/cache/19.07.8/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/debian/asu/worker2/cache/19.07.8/ath79/generic/dl/openwrt_routing
Downloading http://downloads.cdn.openwrt.org/releases/19.07.8/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /home/debian/asu/worker2/cache/19.07.8/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/debian/asu/worker2/cache/19.07.8/ath79/generic/dl/openwrt_telephony
Downloading file:packages/Packages
Updated list of available packages in /home/debian/asu/worker2/cache/19.07.8/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/debian/asu/worker2/cache/19.07.8/ath79/generic/dl/imagebuilder
Downloading https://feed.libremesh.org/master/Packages.gz
Updated list of available packages in /home/debian/asu/worker2/cache/19.07.8/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/debian/asu/worker2/cache/19.07.8/ath79/generic/dl/lime-packages
Downloading https://feed.libremesh.org/profiles/Packages.gz
Updated list of available packages in /home/debian/asu/worker2/cache/19.07.8/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/debian/asu/worker2/cache/19.07.8/ath79/generic/dl/lime-profiles
Collected errors:
 * check_data_file_clashes: Package iw wants to install file /home/debian/asu/worker2/cache/19.07.8/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/usr/sbin/iw
	But that file is already provided by package  * iw-full
 * opkg_install_cmd: Cannot install package iw.
make[2]: *** [Makefile:158: package_install] Error 255
make[1]: *** [Makefile:128: _call_manifest] Error 2
make: *** [Makefile:218: manifest] Error 2
```

CAFxX, 2021-09-26:
Finally, if I remove both `libubus20191227` and `iw` I get a successful build:

[image]

jow-, 2021-09-27:
@aparcar

aparcar, 2021-09-29:
This should be fixed via https://github.com/aparcar/asu/commit/bfecd216f362766cde3dfaa137fb0043e0c8bb79

I'm not sure how to handle the `iw` issue, I think this is a deeper issue with the `opkg` package manager.

aparcar, 2021-09-29:
@jow- if you have any ideas on that matter please let me know.

CAFxX, 2021-10-03:
Just checking in to report that, after upgrading to `git-21.271.68289-79af897-1`, the `libubus` problem is gone.

The `iw` issue is still present.

aparcar, 2021-10-03:
@CAFxX the `iw` issue is an upstream problem fixed via [this opkg commit](https://git.openwrt.org/?p=project/opkg-lede.git;a=commitdiff;h=5936c4f9660248284e8a9b040ea3153d3ea888de;hp=d3a63b327383a2ac1a8f925a58a2f6a7808f5181) in later releases