openwrt/luci #5393: luci-app-wireguard: RPCError RPC call to luci.wireguard/getWgInstances failed with error -32000

teqlee, 2021-09-26:
RPCError RPC call to luci.wireguard/getWgInstances failed with error -32000: Object not found at ClassConstructor.handleCallReply (https://192.168.16.1/luci-static/resources/rpc.js?v=git-21.267.65414-1d9067b:11:3)

Openwrt SNAPSHOT r17584

@lvoegl

feckert, 2021-09-27:
Should be fixed with https://github.com/openwrt/luci/commit/9d560c46cc0663a80d271e2e2dddc5243f6ccfd7

mydogteddy, 2021-09-28:
Not fixed,
Hostname	OpenWrt
Model	Linksys E8450 (UBI)
Architecture	ARMv8 Processor rev 4
Target Platform	mediatek/mt7622
Firmware Version	OpenWrt SNAPSHOT r17630-6cd54254e4 / LuCI Master git-21.226.86205-376af36
Kernel Version	5.10.64

Still seeing 
RPC call to luci.wireguard/getWgInstances failed with error -32000: Object not found

I have clear my cache and checked
luci-proto-wireguard 
file "/usr/libexec/rpcd/luci.wireguard
are installed

teqlee, 2021-09-28:
Still not fixed. Minor difference error showing:
`RPC call to luci.wireguard/getWgInstances failed with error -32000: Object not found
  at handleCallReply (https://192.168.16.1/luci-static/resources/rpc.js?v=git-21.267.65414-1d9067b:11:3)`

feckert, 2021-09-28:
Have you update luci-app-wireguard and luci-proto-wireguard.
Also you have to restart rpcd with the command `/etc/init.d/rpcd `?

mydogteddy, 2021-09-28:
Prior to recompiling the code I did,

./scripts/feeds update -a; ./scripts/feeds install -a

and

git pull --ff-only

then after compiling I flashed the new code, unfortunately the error was still showing

mydogteddy, 2021-09-28:
I have also tried deleting the build_dir and then recompiling it made no difference the error was still there

teqlee, 2021-09-28:
Just to be sure, I rebuilt from scratch by following:
- git pull && ./scripts/feeds update -a
- make package/clean
- make
And i'm sysyupgrade fw without keep config.
my fw infos:
- LuCI Master (git-21.267.65414-1d9067b)
- OpenWrt SNAPSHOT r17631-9c3b1d5563
- luci-app-wireguard - git-21.269.42646-9d560c4
- luci-proto-wireguard - git-21.269.42646-9d560c4`

dhcpme, 2021-09-29:
I can also confirm this is not fixed with 9d560c4 on a Belkin RT3200 (Linksys E8450 (UBI))
[image]
[image]

feckert, 2021-09-29:
What is your output of ` ubus -v list luci.wireguard`

dhcpme, 2021-09-29:
Empty :(
```
root@OpenWrt:~# ubus -v list luci.wireguard
Command failed: Not found
```

mydogteddy, 2021-09-29:
root@OpenWrt:~# ubus -v list luci.wireguard
Command failed: Not found
root@OpenWrt:~#

feckert, 2021-09-29:
It's very strange that it's included in the repository.
Does this [file](https://github.com/openwrt/luci/blob/master/protocols/luci-proto-wireguard/root/usr/libexec/rpcd/luci.wireguard) exist on your target? `/usr/libexec/rpcd/luci.wireguard`

mydogteddy, 2021-09-29:
yes

feckert, 2021-09-29:
@mydogteddy ok could you please restart ubus with following command `/etc/init.d/rpcd restart` and show me the output
of the rpcd with `ubus call luci.wireguard getWgInstances`

dhcpme, 2021-09-29:
```
root@OpenWrt:~# /etc/init.d/rpcd restart
root@OpenWrt:~# ubus call luci.wireguard getWgInstances
Command failed: Not found
```

feckert, 2021-09-29:
@dhcpme See my comment https://github.com/openwrt/luci/issues/5393#issuecomment-930472030

dhcpme, 2021-09-29:
```
root@OpenWrt:~# ls -ltra /usr/libexec/rpcd/luci.wireguard
-rwxr-xr-x    1 root     root          4432 Sep 28 08:27 /usr/libexec/rpcd/luci.wireguard
root@OpenWrt:~# md5sum /usr/libexec/rpcd/luci.wireguard
ae2f10cdcfffe03283680a53728111c6  /usr/libexec/rpcd/luci.wireguard
```

feckert, 2021-09-29:
Sorry I can't reproduce this.
It works for me?
[image]

```
root@st-dev-07 /usr/libexec/rpcd # md5sum luci.wireguard
ae2f10cdcfffe03283680a53728111c6  luci.wireguard
```

mydogteddy, 2021-09-29:
I am glad it works for you

dhcpme, 2021-09-29:
Related?
https://github.com/openwrt/luci/issues/4555#issuecomment-927104087

mydogteddy, 2021-09-29:
root@OpenWrt:/usr/libexec/rpcd# md5sum luci.wireguard
ae2f10cdcfffe03283680a53728111c6  luci.wireguard
root@OpenWrt:/usr/libexec/rpcd#

feckert, 2021-09-29:
@mydogteddy @dhcpme and @teqlee
Could you please check if all files from von [luci-app-wireguard](https://github.com/openwrt/luci/tree/master/applications/luci-app-wireguard) and [luci-proto-wireguard](https://github.com/openwrt/luci/tree/master/protocols/luci-proto-wireguard) are installed on your system an does have the content as in the luci (md5sum check)?

We're talking about the master branch?

teqlee, 2021-09-29:
Here md5sum on my system, I dont know how compare to master repos.

`md5deep -r ~/opentrunk/feeds/luci/applications/luci-app-wireguard
083bbe067553ffcae725f72716b72471  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/ar/wireguard.po
4d9d8fca4460de49a0f93f0ffe92dfb6  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/mr/wireguard.po
0a8377e7f8dc8e7b5631f3dcd7b33d66  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/zh_Hant/wireguard.po
1ebd900a3033174704f06cd183ab08c3  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/it/wireguard.po
dbd8770ac9bc92cf867150972dcc2332  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/vi/wireguard.po
4287e7af22941125b41e55d9f3eaef67  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/ru/wireguard.po
99ef98f414cd67d3eb6a4e00a51beead  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/el/wireguard.po
8ae901c6cf0907f5e9abffef361b36ba  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/sv/wireguard.po
8d4dba2779d8822eb350273e44bc041e  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/fi/wireguard.po
28b2a755cb0d214014439c389f59f00f  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/ja/wireguard.po
ba8b2ea7f5397fc6941c88175ca8d8b9  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/zh_Hans/wireguard.po
0273eaa8e6d9c7d184544add166b0498  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/pl/wireguard.po
1617e1f889aca223915308c407db0171  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/pt/wireguard.po
861dd73e6e6700081c785b16d10d033e  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/cs/wireguard.po
a4856a40b29ea9096480edc1cfbabbec  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/he/wireguard.po
0bb07f6371e3e3c42d60cb697b2a8ea1  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/nb_NO/wireguard.po
26e16a7a51c9a62e2be32966ee94db38  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/bn_BD/wireguard.po
e2d6313ec4e9fd3eaa5ff90f6b506bb5  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/templates/wireguard.pot
27530c98ca92307649dafac664dafbd5  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/sk/wireguard.po
a4b32c3edb998b42a412c7e60b507703  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/ro/wireguard.po
ec84e2370ca7b37a6c1d74cf41df44ff  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/hu/wireguard.po
b7d548f9f8563f603db4d90483bd6631  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/fr/wireguard.po
fb52672bb64bddf557d58cb10d3783cd  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/ko/wireguard.po
e95e83ba76b257c1758fa531c3d03465  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/tr/wireguard.po
17b7a11e237d5c8b88591d2e72b42248  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/en/wireguard.po
300935669ca05219d94507329bc91aee  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/uk/wireguard.po
7c216d87a70266ecda82d074c9da8f65  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/hi/wireguard.po
7d3ba5761095c7803c5c364bbcc55b69  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/pt_BR/wireguard.po
6c107c1ce964640a8ed6bb6af4ae924d  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/bg/wireguard.po
924b8305b4924f6aceea476be542b5e5  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/de/wireguard.po
e80151f33ba58357f4c97f8ed749dcf3  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/id/wireguard.po
352bc6a36ce6cf7b869e7e2e6ed8ea8a  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/ms/wireguard.po
b22d6043ebe6fee78ca8bc6d4cf6630b  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/es/wireguard.po
1e5135b8b71995cfc6f8b8ea1d47ff27  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/po/ca/wireguard.po
9a1e9c01566472b087e0d0192c9df267  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/root/usr/share/luci/menu.d/luci-app-wireguard.json
718474b39bfb6e597e3d5a12dfb5a7a5  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/root/usr/share/rpcd/acl.d/luci-app-wireguard.json
ecc5deaeed18e5602d6e8f0993632753  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/htdocs/luci-static/resources/view/wireguard/status.js
c71ec3e475714afa21e1bbe4c66732e0  /home/teq/opentrunk/feeds/luci/applications/luci-app-wireguard/Makefile

md5deep -r ~/opentrunk/feeds/luci/protocols/luci-proto-wireguard
ae2f10cdcfffe03283680a53728111c6  /home/teq/opentrunk/feeds/luci/protocols/luci-proto-wireguard/root/usr/libexec/rpcd/luci.wireguard
c86175d91113469d1ad66c4c63950bdb  /home/teq/opentrunk/feeds/luci/protocols/luci-proto-wireguard/Makefile
43fcd70d409ec70c149d6809d00ebef9  /home/teq/opentrunk/feeds/luci/protocols/luci-proto-wireguard/root/usr/share/rpcd/acl.d/luci-wireguard.json
78788cb08a2ee4da898699ef1563bdeb  /home/teq/opentrunk/feeds/luci/protocols/luci-proto-wireguard/htdocs/luci-static/resources/protocol/wireguard.js`

teqlee, 2021-09-29:
I'm trying git clone https://git.openwrt.org/project/luci.git
check md5sum and it's the same:
```
teq@TeqDesk:~/downloads$ md5deep -r luci/applications/luci-app-wireguard
9a1e9c01566472b087e0d0192c9df267  /home/teq/downloads/luci/applications/luci-app-wireguard/root/usr/share/luci/menu.d/luci-app-wireguard.json
718474b39bfb6e597e3d5a12dfb5a7a5  /home/teq/downloads/luci/applications/luci-app-wireguard/root/usr/share/rpcd/acl.d/luci-app-wireguard.json
ecc5deaeed18e5602d6e8f0993632753  /home/teq/downloads/luci/applications/luci-app-wireguard/htdocs/luci-static/resources/view/wireguard/status.js
c71ec3e475714afa21e1bbe4c66732e0  /home/teq/downloads/luci/applications/luci-app-wireguard/Makefile

teq@TeqDesk:~/downloads$ md5deep -r luci/protocols/luci-proto-wireguard
ae2f10cdcfffe03283680a53728111c6  /home/teq/downloads/luci/protocols/luci-proto-wireguard/root/usr/libexec/rpcd/luci.wireguard
43fcd70d409ec70c149d6809d00ebef9  /home/teq/downloads/luci/protocols/luci-proto-wireguard/root/usr/share/rpcd/acl.d/luci-wireguard.json
78788cb08a2ee4da898699ef1563bdeb  /home/teq/downloads/luci/protocols/luci-proto-wireguard/htdocs/luci-static/resources/protocol/wireguard.js
c86175d91113469d1ad66c4c63950bdb  /home/teq/downloads/luci/protocols/luci-proto-wireguard/Makefile

feckert, 2021-09-29:
Yes, I can confirm that I have the same hashes.
Have you cleared your browser cache?
But this would not explain this error result https://github.com/openwrt/luci/issues/5393#issuecomment-930489898

teqlee, 2021-09-29:
Yep, tried cleared all cache, disabled all add-on/extensions, also trying Firefox Chrome Edge, the same.

castillofrancodamian, 2021-09-30:
The button to generate a private key does not work either.

jow-, 2021-09-30:
Please execute `/usr/libexec/rpcd list` as command on the cli and see if it crashes. I suspect it requires one or more LuCI libraries which are not listed as dependencies.

mydogteddy, 2021-09-30:
root@OpenWrt:~# /usr/libexec/rpcd/luci.wirguard list
-ash: /usr/libexec/rpcd/luci.wirguard: not found
root@OpenWrt:~#

mydogteddy, 2021-09-30:
there is a typo in jows command
root@OpenWrt:~# /usr/libexec/rpcd/luci.wirguard list
-ash: /usr/libexec/rpcd/luci.wirguard: not found
root@OpenWrt:~# git checkout
-ash: git: not found
root@OpenWrt:~# /usr/libexec/rpcd/luci.wirguard list
-ash: /usr/libexec/rpcd/luci.wirguard: not found
root@OpenWrt:~# /usr/libexec/rpcd/luci.wireguard list
lua: /usr/libexec/rpcd/luci.wireguard:6: module 'uci' not found:
        no field package.preload['uci']
        no file './uci.lua'
        no file '/usr/share/lua/uci.lua'
        no file '/usr/share/lua/uci/init.lua'
        no file '/usr/lib/lua/uci.lua'
        no file '/usr/lib/lua/uci/init.lua'
        no file './uci.so'
        no file '/usr/lib/lua/uci.so'
        no file '/usr/lib/lua/loadall.so'
stack traceback:
        [C]: in function 'require'
        /usr/libexec/rpcd/luci.wireguard:6: in main chunk
        [C]: ?
root@OpenWrt:~#

feckert, 2021-09-30:
As already mentioned by @jow- (thanks for this) there is a dependency issue with luci.wireguard that I have not seen on merge

All of you have to install libuci-lua with the command `opkg update && opkg install libuci-lua`
I will prepare a fix for this of this is the solution

mydogteddy, 2021-09-30:
I compiled again including libuci-lua but it did not correct the issue.

root@OpenWrt:~# opkg list_installed | grep libuci
libuci20130104 - 2021-04-14-4b3db117-6
root@OpenWrt:~#

                                  -*- libuci........... C library for the Unified Configuration Interface (UCI)                            │ │  
  │ │                                  <*> libuci-lua............................................ Lua plugin for UCI                            │ 

root@OpenWrt:~# /usr/libexec/rpcd/luci.wireguard list
lua: /usr/libexec/rpcd/luci.wireguard:6: module 'uci' not found:
        no field package.preload['uci']
        no file './uci.lua'
        no file '/usr/share/lua/uci.lua'
        no file '/usr/share/lua/uci/init.lua'
        no file '/usr/lib/lua/uci.lua'
        no file '/usr/lib/lua/uci/init.lua'
        no file './uci.so'
        no file '/usr/lib/lua/uci.so'
        no file '/usr/lib/lua/loadall.so'
stack traceback:
        [C]: in function 'require'
        /usr/libexec/rpcd/luci.wireguard:6: in main chunk
        [C]: ?
root@OpenWrt:~#

mydogteddy, 2021-09-30:
I compiled and flashed again just to be sure and now I see this

root@OpenWrt:~# /usr/libexec/rpcd/luci.wireguard list
{"generateQrCode":{"allowed_ips":["allowed_ips"],"privkey":"privkey","psk":"psk"},"generateKeyPair":{},"getWgInstances":{}}
root@OpenWrt:~#

feckert, 2021-09-30:
That looks good, then try the LuCI to see if it fixes the problem now.

mydogteddy, 2021-09-30:
In luci I see this,
https://imgur.com/mwnjNLZ.png

I have a WG running from my phone to the router at the same time

mydogteddy, 2021-09-30:
wg show has all the correct info

jow-, 2021-09-30:
> In luci I see this,

Possibly a theme (XHTML) specific issue, please retest with the default bootstrap theme.

mydogteddy, 2021-09-30:
https://imgur.com/vGz9iCI.png

mydogteddy, 2021-09-30:
I can confirm OpenWrt | OpenWrt SNAPSHOT r17531-f413e4835e | Load: 0.54 0.18 0.06  works OK if this is of any use

mydogteddy, 2021-09-30:
This version works OK

Hostname	OpenWrt
Model	Linksys E8450 (UBI)
Architecture	ARMv8 Processor rev 4
Target Platform	mediatek/mt7622
Firmware Version	OpenWrt SNAPSHOT r17531-f413e4835e / LuCI Master git-21.226.86205-376af36
Kernel Version	5.10.64
Local Time	2021-09-30 10:35:17
Uptime	0h 3m 12s
Load Average	0.05, 0.11, 0.05

mydogteddy, 2021-09-30:
this version does not work OK

Hostname	OpenWrt
Model	Linksys E8450 (UBI)
Architecture	ARMv8 Processor rev 4
Target Platform	mediatek/mt7622
Firmware Version	OpenWrt SNAPSHOT r17630-6cd54254e4 / LuCI Master git-21.226.86205-376af36
Kernel Version	5.10.64
Local Time	2021-09-30 10:39:31
Uptime	0h 2m 17s
Load Average	0.28, 0.16, 0.06

mydogteddy, 2021-09-30:
for clarity I have 2 x these Belkins one with the code I have recently compiled (which does not show in luci the WG status) and  the other one which has some code compiled a short while ago and does show the WG status in luci

teqlee, 2021-09-30:
Installed libuci-lua:

[image]

castillofrancodamian, 2021-09-30:
> Installed libuci-lua & rebooted:
> 
> [image]
> 
> Also, generate priv-key button comeback to work after install libuci-lua.

The same here. Even with this dependency installed, the WireGuard status does not appear.

mydogteddy, 2021-09-30:
I have just done a new git clone however WG status is still not working

dhcpme, 2021-09-30:
I installed the 'libuci-lua' dependency before using attended-sysupgrade to install a new image with it included and the status page appeared to work for a while but resembled the screenshots above by the time I connected a peer. I lost the ability to login to Luci after a reboot so I reverted to an image made yesterday via the CLI although it may have been my browser. I may try again later or sit back for a week and let you guys ride this out. :o)

lvoegl, 2021-10-01:
> > Installed libuci-lua & rebooted:
> > [image]
> > Also, generate priv-key button comeback to work after install libuci-lua.
> 
> The same here. Even with this dependency installed, the WireGuard status does not appear.

[image]

Do you have any servers configured? The WG Status does look like this without any configurations.

mydogteddy, 2021-10-01:
There was less than 100 commits between WG status working and it not working  
Firmware Version OpenWrt SNAPSHOT r17531-f413e4835e / LuCI Master git-21.226.86205-376af36
Firmware Version OpenWrt SNAPSHOT r17630-6cd54254e4 / LuCI Master git-21.226.86205-376af36

teqlee, 2021-10-01:
> Do you have any servers configured? This is the old vs. new Status with and without a Wireguard configuration.

Yes, i have server & client wg configured.
[image]
```
config interface 'wg1'
        option proto 'wireguard'
        option private_key 'xxx'
        option metric '9'
        option delegate '0'
        option peerdns '0'
        option mtu '1280'
        list addresses 'xxxx'

config wireguard_wg1
        option public_key 'xxxx'
        option endpoint_port '2408'
        option persistent_keepalive '25'
        option route_allowed_ips '1'
        option description 'Cloudflare'
        option endpoint_host 'xxxx'
        list allowed_ips '0.0.0.0/0'

config interface 'wg0'
        option proto 'wireguard'
        option delegate '0'
        option private_key 'xxxx'
        list addresses '10.111.111.1/24'
        option force_link '1'
        option listen_port '55820'
        option peerdns '0'

config wireguard_wg0
        option route_allowed_ips '1'
        option persistent_keepalive '25'
        list allowed_ips '10.111.111.5/32'
        option description 'x'
        option public_key 'xxxx'

lvoegl, 2021-10-01:
should be fixed with #5403

teqlee, 2021-10-01:
> 
> 
> should be fixed with #5403

Well, confirm fixed. Tks for hard work.

feckert, 2021-10-01:
Thanks all for testing
@lvoegl thanks for fixing

PR #5403 (2021-10-01): luci-app-wireguard: fix allowed_ip parsing
fix #5393