openwrt/luci #5394: Wireless ieee80211w with WPA3

ByCzech, 2021-09-26:
802.11w Management Frame Protection option has 3 values: disabled, optional and required.

Luci sets it with:

"uci del wireless.default_radioN.ieee80211w" for "disabled"
"uci set wireless.default_radioN.ieee80211w='1'" for "optional"
"uci set wireless.default_radioN.ieee80211w='2'" for "required"

which is OK for WPA and WPA2 modes. But in WPA2/WPA3 mixed mode or in WPA3 mode "disabled" not working. For disabling 802.11w is needed manual setting:

uci set wireless.default_radioN.ieee80211w='0'

this option setting is working well as "disabled" for WPA and WPA2 modes too. So I wish to Luci change his behavior in this setting.

hnyman, 2021-09-27:
to my understanding, ieee80211w is needed for WPA3, so you can't have WPA3 (or mixed WPA2/WPA3) and disable ieee80211w by setting it to 0

closed 2021-09-27 by jow-

ByCzech, 2021-09-27:
I have devices, which need to work in WPA2 (WPA2+WPA3 mixed mode). If ieee80211w can't be disabled, this devices don't work. If I disabled it manualy, this devices works and devices capable WPA3 works too.

marcluer, 2021-09-27:
I have the same problem too.
My Raspi 3 is not able to connect to my OpenWrt router, as long as ieee80211w is set to "optional" or "required".
I need to set ieee80211w='0' manually, which I can´t do through the webinterface.

@jow- If this is against the standard, the dropdown option "disabled" should not be visible in the first place

ByCzech, 2021-09-28:
> @hnyman If this is against the standard, the dropdown option "disabled" should not be visible in the first place.

Or add new "default" option is needed.
If hw/sw can do "disabled" although it is against the standard, because users need it and want it, then isn't reason to obstruct this in GUI.

blogic, 2021-09-28:
just following this on the side ... mac80211.sh will override whatever uci has set for 80211w when wpa3 is set and use 2 (required)

marcluer, 2021-09-28:
@ByCzech I just noticed, that with "disabled" my Raspberry Pi works, but my Sony Smartphone stopped working.
What actual hardware are you using?

ByCzech, 2021-09-29:
> @ByCzech I just noticed, that with "disabled" my Raspberry Pi works, but my Sony Smartphone stopped working. What actual hardware are you using? I am going back to WPA2 for now.

Few  elderly mobile phones and one notebook with Broadcom chip, which need compiling wl kernel module, for to be fully operational in 2.4 and 5 GHz bands.

commit 0b49ed45 (2021-09-27): luci-mod-network: fix disabling 11w MFP for WPA3

Fixes: #5394