openwrt/luci #5407: Luci not able to edit WireGuard interface

mydogteddy, 2021-10-05:
Hostname	SKY1
Model	Linksys E8450 (UBI)
Architecture	ARMv8 Processor rev 4
Target Platform	mediatek/mt7622
Firmware Version	OpenWrt SNAPSHOT r17675-ed7769aa40 / LuCI Master git-21.226.86205-376af36
Kernel Version	5.10.70

Luci shows a SyntaxError when trying to edit the WireGuard interface

An invalid or illegal string was specified

feckert, 2021-10-06:
Who does your uci network config for wireguard look like?

mydogteddy, 2021-10-06:
This is my WG config keys changed of course.
config interface 'wg0'
	option proto 'wireguard'
	option listen_port '51820'
	option private_key 'AIhSzGTTHPaqof7esgwuqorpNq4lHQFBoCB5ZJn5MWk='
	list addresses '192.168.0.0/22'
	option metric '20'

config wireguard_wg0 'wgclient1'
	option description 'My note3'
	option route_allowed_ips '1'
	option public_key 'TuaZkz+zK32+QrvGv4DRiemv3XtLVtlH6qopd8Hr3Q8='
	option preshared_key '9cDggBYPahdyencis9n8WISl3Vpqhairhes7f030rbY='
	option endpoint_host 'xxxxxxxx.ddns.ms'
	option endpoint_port '51820'
	list allowed_ips '192.168.9.23/32'

config wireguard_wg0 'wgclient2'
	option description 'Laptop'
	option public_key 'TeO2d9NJUHqpZKzK09wVc3aPjPpGkBIbCjb7AAUr/2k='
	option preshared_key '9cDggBYPahdyencis9n8WISl3Vpqhairhes7f030rbY='
	option route_allowed_ips '1'
	option endpoint_host 'xxxxxxl.ddns.ms'
	option endpoint_port '51820'
	list allowed_ips '192.168.10.23/32'

closed 2021-10-07 by jow-