openwrt/luci #5474: luci-app-softether does not show any information

NilsRo, 2021-10-30:
Luci App for Softether does not show any information and hangs on "Loading account information…". Softether itself runs very fine.

[image]

Using OpenWrt 21.02.0 r16279-5cc0535800 / LuCI openwrt-21.02 branch git-21.285.75922-4fd8c83

jow-, 2021-10-31:
Please open your browser debug console with F12 and look for errors

NilsRo, 2021-10-31:
Yes, there is a 403 error visible which matches the plugin.

[image]

jow-, 2021-10-31:
Did you reboot or logged out/logged in at least once since you installed the app?

NilsRo, 2021-11-01:
Yes. Found the issue. I installed vpn bridge as well as there are more options for configuration. vpn client is also installed but I switched to bridge for flexibillity. So an error in front of the monitor. :-) But would be nice if vpn bridge will be also supported. Perhaps it is easy to add...

jow-, 2021-11-02:
Okay, so the issue is that you already uninstalled the package again but the menu was still displaying the entry?

NilsRo, 2021-11-02:
If I stop Softether Bridge and start Softether Client it is working fine. I saw this as only Softether Client is listed as dependency. Will perhaps think about switching to Client but like the Softether Bridge as it can be configured like a hub and also this is more the case with OpenWRT Router.

Mohsen7s, 2021-11-26:
I also have same issue, here is what I tried.
Modem Factory Reset.
A fresh install of luci-app-softether which also install softethervpn5 client as dependency.
Once I go to softether management URL, I get exactly permission deny by ACL.

ringtailedfox, 2022-05-22:
is there a way to allow access to these resources so luci/openwrt doesn't whine about HTTP 403: Permission Denied errors?

asmwarrior, 2023-06-01:
> 

I have the same issue here. 
When I click on the F12, I see some error message like: PermissionError: Access to command denied by ACL
I just installed the `luci-app-softether`. I have reboot the system, also, I have logged out and logged in again.

Any one know how to solved this issue? Thanks.

nikemir, 2024-03-13:
The error stems from an hanging, unattended password prompt in accessing the VPN client services.

I traced the error all the way to the 'local' terminal command that is being executed.
Note that the API call from luci is remote but the terminal command being executed is 'local'.

You will not encounter the issue if no password is set as it bypasses password prompts entirely.
If you have/need a password set, must set /REMOTEONLY:yes to bypass password prompts for local operations.

Problem:
VPN Client>PasswordGet
Set Password | Enable
Request Password for Remote Operation Only | Disable
_(Require password for all operations)_

Solution 1:
VPN Client>PasswordGet
Set Password | Enable
Request Password for Remote Operation Only | Enable
_(Bypass password for 'local' operations)_
To achieve, use:
`VPN Client>PasswordSet <YourPassword> /REMOTEONLY:yes`

Solution 2:
VPN Client>PasswordGet
Set Password | Disable
Request Password for Remote Operation Only | Disable
_(Bypass password prompt entirely)_
To achieve, use:
`VPN Client>PasswordSet none`

Got the SoftEther Status page working.
It is not in any way a configuration panel, more of an informative view of the client services.
Good luck!

ringtailedfox, 2024-03-15:
is this done in OpenWRT? or do i need a VPN Program in Windows/Linux like SoftEther VPN client?

nikemir, 2024-03-15:
Yes in OpenWRT.
You have to ssh in or use terminal in OpenWRT and run:
vpncmd > 2 (for vpn client) > localhost (for destination)

Then you will be presented with the VPN Client> prompt where you can use the commands above like PasswordGet and PasswordSet.

Just so that we’re on the same page, the panel is to display the connection status as a client. I have my router here (se client) to connect to an office router (se server) so that all my devices here can access office resources, without setting up/configuring each one of them. I had a previous misunderstanding where I would see the pool of clients connected to a server.