openwrt/luci #5476: Luci Led Configuration add and edit broken in snapshots

CaboBeach, 2021-10-31:
When clicking on Add LED action or the Edit button for a current led there's just a spinning wheel over the button and no pop up dialog box appears to add or edit, works fine in cli and the leds appear in led configuration after adding/editing in cli.

[image]

[image]

jow-, 2021-10-31:
Please open your browser debug console with `F12` and look for errors.

closed 2021-10-31 by jow-

CaboBeach, 2021-10-31:
Here's the debug console, it was the same for both add and edit - 

[image]

GqSteve, 2021-11-01:
How can I apply this fix?

Copy the 5 files from Git to /www/luci-static/resources/view/system/led-trigger ?

jow-, 2021-11-01:
You only need to update the `none.js` file.

GqSteve, 2021-11-01:
I "updated" the content of https://raw.githubusercontent.com/openwrt/luci/fceabd74465f2aafc40edc084429f30837deff2c/modules/luci-mod-system/htdocs/luci-static/resources/view/system/led-trigger/none.js in the  /www/luci-static/resources/view/system/led-trigger/none.js, but I still have the same behavior as described in #1: .._.there's just a spinning wheel over the button and no pop up dialog box appears to add or edit..._

I have 21.02.1 installed on Fritz!Repeater 3000. Should I open a new issue?

GqSteve, 2021-11-02:
I updated all files from fix in the  /www/luci-static/resources/view/system/led-trigger/ directory and worked fine. Thank you! 👍

feckert, 2021-11-02:
@jow- Thank you for looking at this. It worked for me in my integration test. Very strange.

commit fceabd74 (2021-10-31): luci-mod-system: add missing require to `none` led trigger class

Also convert ES6 to ES5 syntax while we're at it.

Fixes: #5476
Fixes: 5b42cd5b46 ("luci-mod-system: move default option from defaul-on trigger to none trigger")
commit 388dae96 (2021-11-01): luci-mod-system: add missing require to `none` led trigger class

Also convert ES6 to ES5 syntax while we're at it.

Fixes: #5476
Fixes: 5b42cd5b46 ("luci-mod-system: move default option from defaul-on trigger to none trigger")

(cherry picked from commit fceabd74465f2aafc40edc084429f30837deff2c)