openwrt/luci #5477: luci-mod-system/startup: dnsmasq restart failed to load names from /tmp/hosts/dhcp.main_dns: Permission denied

rhuang2014, 2021-10-31:
Dnsmasq no longer able to read dns files in /tmp/hosts/ after restart via the UI.  All files are `rw` by owner (root) only.
```shell
root@u1:/etc/config# ls -l /tmp/hosts/dhcp.main_dns 
-rw-------    1 root     root           151 Oct 30 23:53 /tmp/hosts/dhcp.main_dns
root@u1:/etc/config# logread -e dnsmasq.*dhcp.main_dns
Sat Oct 30 23:53:16 2021 daemon.err dnsmasq[11280]: failed to load names from /tmp/hosts/dhcp.main_dns: Permission denied
```

hnyman, 2021-10-31:
cc @dangowrt @jow- 

Confirmed.
I tested restarting dnsmasq via LuCI, and you are right:
The permissions of the dhcp hosts file get changed from 644 to 600 in the restart by LuCI, while restarting via SSH console then returns it to the proper 644.

Why does restart via LuCI cause different file permissions for a temp config file generated by the init script?

I first thought that this might be an effect from adopting ujail by default, but the file permission thing does not relate to it, right?

I also tested restarting luci_statistics & collectd, as well as hostapd(wpad) that also runs with a different user but their temp config files still have the proper file permissions as a restart via LuCI.

Below is the debug evidence for dnsmasq:

```
root@router1:~# ls -l /tmp/hosts/
-rw-r--r--    1 root     root           130 Oct 31 00:03 dhcp.cfg01411c
-rw-r--r--    1 root     root           239 Oct 31 08:58 odhcpd

*** dnsmasq restart via LuCI system/startup
root@router1:~# ls -l /tmp/hosts/
-rw-------    1 root     root           130 Oct 31 09:04 dhcp.cfg01411c
-rw-r--r--    1 root     root           239 Oct 31 08:58 odhcpd

root@router1:~# /etc/init.d/dnsmasq restart
root@router1:~# ls -l /tmp/hosts/
-rw-r--r--    1 root     root           130 Oct 31 09:04 dhcp.cfg01411c
-rw-r--r--    1 root     root           239 Oct 31 08:58 odhcpd
```

PS.
regarding ujail:
should  `/tmp/hosts` dir (maybe as `$(dirname $HOSTFILE)` ) be added to the ujail list in https://github.com/openwrt/openwrt/blob/master/package/network/services/dnsmasq/files/dnsmasq.init#L1152-L1157

Actually, most of the (static) leased device name are missing from that file, although the leases are active. There are just two addressed for the router itself, plus curiously one other address.

Might be that there is something strange going on with dnsmasq, unrelated to LuCI.

dangowrt, 2021-10-31:
> Why does restart via LuCI cause different file permissions for a temp config file generated by the init script?

I'd guess `umask` is modified by LuCI and not touched by the init script.

> should /tmp/hosts dir (maybe as $(dirname $HOSTFILE) )

As long as `$(dirname $HOSTFILE) != /tmp`, yes.

hnyman, 2021-10-31:
@dangowrt 

You can hopefully take care of package/network/services/dnsmasq/files/dnsmasq.init, as that part is not LuCI related, but ujail.

>  As long as $(dirname $HOSTFILE) != /tmp, yes.

It is /tmp/hosts

dnsmasq init file is a bit convoluted...

Key lines:

https://github.com/openwrt/openwrt/blob/master/package/network/services/dnsmasq/files/dnsmasq.init#L15
`BASEHOSTFILE="/tmp/hosts/dhcp"`

https://github.com/openwrt/openwrt/blob/master/package/network/services/dnsmasq/files/dnsmasq.init#L830-L837
```
HOSTFILE="${BASEHOSTFILE}.${cfg}"
HOSTFILE_TMP="${HOSTFILE}.$$"
...
mkdir -p $(dirname $HOSTFILE)
```

https://github.com/openwrt/openwrt/blob/master/package/network/services/dnsmasq/files/dnsmasq.init#L1132
`mv -f $HOSTFILE_TMP $HOSTFILE`

ujail block where the dir is missing:
https://github.com/openwrt/openwrt/blob/master/package/network/services/dnsmasq/files/dnsmasq.init#L1152-L1157
```
	procd_add_jail dnsmasq ubus log
	procd_add_jail_mount $CONFIGFILE $DHCPBOGUSHOSTNAMEFILE $DHCPSCRIPT $DHCPSCRIPT_DEPENDS
	procd_add_jail_mount $EXTRA_MOUNT $RFC6761FILE $TRUSTANCHORSFILE
	procd_add_jail_mount $dnsmasqconffile $dnsmasqconfdir $resolvdir $user_dhcpscript
	procd_add_jail_mount /etc/passwd /etc/group /etc/TZ /etc/hosts /etc/ethers
	procd_add_jail_mount_rw /var/run/dnsmasq/ $leasefile
```

Note that the directory also contains the odhcpd file, so permissions for a single file is not enough.

dangowrt, 2021-10-31:
Please check if the patch below resolves both issues (wrong umask inherited from LuCI, missing HOSTFILE_DIR in ujail).

```diff
diff --git a/package/network/services/dnsmasq/files/dnsmasq.init b/package/network/services/dnsmasq/files/dnsmasq.init
index 3250b2179b..28cc8b961b 100644
--- a/package/network/services/dnsmasq/files/dnsmasq.init
+++ b/package/network/services/dnsmasq/files/dnsmasq.init
@@ -829,12 +829,14 @@ dnsmasq_start()
        CONFIGFILE_TMP="${CONFIGFILE}.$$"
        HOSTFILE="${BASEHOSTFILE}.${cfg}"
        HOSTFILE_TMP="${HOSTFILE}.$$"
+       HOSTFILE_DIR="$(dirname $HOSTFILE)"
        BASEDHCPSTAMPFILE_CFG="${BASEDHCPSTAMPFILE}.${cfg}"
 
        # before we can call xappend
+       umask u=rwx,g=rx,o=rx
        mkdir -p /var/run/dnsmasq/
        mkdir -p $(dirname $CONFIGFILE)
-       mkdir -p $(dirname $HOSTFILE)
+       mkdir -p "$HOSTFILE_DIR"
        mkdir -p /var/lib/misc
        chown dnsmasq:dnsmasq /var/run/dnsmasq
 
@@ -1154,6 +1156,7 @@ dnsmasq_start()
        procd_add_jail_mount $EXTRA_MOUNT $RFC6761FILE $TRUSTANCHORSFILE
        procd_add_jail_mount $dnsmasqconffile $dnsmasqconfdir $resolvdir $user_dhcpscript
        procd_add_jail_mount /etc/passwd /etc/group /etc/TZ /etc/hosts /etc/ethers
+       [ "$HOSTFILE_DIR" != "/tmp" ] && procd_add_jail_mount "$HOSTFILE_DIR"
        procd_add_jail_mount_rw /var/run/dnsmasq/ $leasefile
 
        procd_close_instance
```

hnyman, 2021-10-31:
Thanks.
I will test it.

Ps.
I wonder about your umask syntax "umask u=rwx,g=rx,o=rx".
I think that we are usually using numeric style (022 is the same , I think).

https://github.com/openwrt/openwrt/search?q=umask

dangowrt, 2021-10-31:
I just copy-pasted the output of `umask -S` inside SSH :) But true, it can also be specified numerically.

hnyman, 2021-10-31:
Thanks, I edited manually the init file in the live router, (and used umask 022).

Looks like the permissions are now ok even after a restart via LuCI, and ps shows /tmp/hosts as an allowed dir:

```
21303 root      2392 S    {dnsmasq} /sbin/ujail -n dnsmasq -u -l -r /bin/ubus -r /etc/TZ -r /etc/dnsmasq.conf -r /etc/ethers -r /etc/group -r /etc/hosts -r /etc/passwd -w /tmp/dhcp.leases -r /tmp/dnsmasq.d -r /tmp/hosts -r /tmp/resolv.conf.d ...
21305 dnsmasq   3700 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411c.pid
```

```
root@router1:/tmp#
root@router1:/tmp# ls -l /tmp/hosts/
-rw-r--r--    1 root     root           130 Oct 31 11:58 dhcp.cfg01411c
-rw-r--r--    1 root     root           173 Oct 31 11:45 odhcpd

--- RESTART via LuCI ---

root@router1:/tmp# ls -l /tmp/hosts/
-rw-r--r--    1 root     root           130 Oct 31 12:00 dhcp.cfg01411c
-rw-r--r--    1 root     root           173 Oct 31 11:45 odhcpd
```

dangowrt, 2021-10-31:
Should be fixed by https://github.com/openwrt/openwrt/commit/a2fcd3900c0c7c168c4d3c04da189e408860483d

closed 2021-10-31 by dangowrt