openwrt/luci #5486: Update check or/and OTA update

olekstomek, 2021-11-04:
What do you think about the possibility of checking the availability of the new version in the GUI? It seems very convenient.
Or maybe, in addition to the notification of a new version, it is possible to add an OTA update of firmware?

[This GUI](https://eko.one.pl/?p=mifi) for OpenWrt has this functionality and it works. :)
[image]

jow-, 2021-11-04:
There is no server side infrastructure in OpenWrt yet to support such a feature. If there is eventually, we'll implement checking in LuCI.

closed 2021-11-04 by jow-

mwarning, 2021-11-06:
This sounds like the attended sysupgrade feature. It has also a luci plugin.

olekstomek, 2021-11-07:
> It has also a luci plugin.

Can you expand? Do you mean [luci-app-attendedsysupgrade](https://github.com/openwrt/luci/tree/master/applications/luci-app-attendedsysupgrade)?

Btw Gargoyle has option for checking new version of firmware:
[image]

olekstomek, 2021-11-07:
OK, **luci-app-attendedsysupgrade** works! :)
[image]

`/cgi-bin/luci/admin/system/attendedsysupgrade/overview`

bam80, 2021-11-08:
> There is no server side infrastructure in OpenWrt yet to support such a feature. If there is eventually, we'll implement checking in LuCI.

Do we have an issue or somewhat to track it?

mwarning, 2021-11-09:
@aparcar ping

aparcar, 2021-11-09:
@mwarning this issue seem resolved. Someone could integrate `luci-app-attendedsysupgrade` into the sysupgrade view and ship it per default.

bam80, 2021-11-09:
@aparcar as we have discussed, `luci-app-attendedsysupgrade` has no option yet to flash official builds/check third-party builds for MD5 sums matching.
As such, don't you think that solution is not appropriate to be included by default?

aparcar, 2021-11-09:
If I remember correctly the LuCI app verified the checksum and runs sysupgrade in a way that it verified signatures.

The service is running on OpenWrt infrastructure and uses a domain, what more "official" do you need?

bam80, 2021-11-09:
AFAIK, the LuCI app calculates the uploaded image checksum and suggests user to check it manually before the upgrade.
That check is only possible (make sense) if you upload existing published image.
As OpenWrt builds are not reproducible (yet?), that check will always fail with custom-build images.

We, users, don't have to trust the infrastructure the service is running.
We need firmware checksum matching, to preserve the security and maintainability after upgrade.

aparcar, 2021-11-10:
>  We need firmware checksum matching, to preserve the security and maintainability after upgrade.

The [LuCI app](https://github.com/openwrt/luci/blob/master/applications/luci-app-attendedsysupgrade/htdocs/luci-static/resources/view/attendedsysupgrade/overview.js#L85) as well as [`auc`](https://github.com/openwrt/packages/blob/master/utils/auc/src/auc.c#L1743) verify the checksums provided by the server. If you're installing one of the tools you'll also install [`attendedsysupgrade-common`](https://github.com/openwrt/packages/blob/master/utils/attendedsysupgrade-common/files/86241a707a30cb7f) which adds a usign key. All build images are signed and the signature is tested prior installation.

> We, users, don't have to trust the infrastructure the service is running.

Of course you have to trust the infrastructure. What's worth a checksum if it'd be the checksum of a flawed image? If you don't feel comfortable with `sysupgrade.openwrt.org` please use `downloads.openwrt.org` and build your custom scripts. That's just outside the scope of the service.

bam80, 2021-11-10:
> The LuCI app as well as auc verify the checksums provided by the server.

The only problem the checksums provided won't match those published on `downloads.openwrt.org`.
Even if the infrastructure is the same, I see it as serious flaw.

I appreciate your work on the service, it's step forward anyway, just shared my thoughts how it could be complimented.
If it's out of scope then it should be provided in other service, maybe. As we need it anyway..

aparcar, 2021-11-10:
I don't think it's needed. As I previously explained you could help working on the reproducibility because that would result in the very same images.

bam80, 2021-11-10:
Do we have issues opened why it's not reproducible right now?
Do we know places where it breaks?