openwrt/luci #5491: jsmin.c is nonfree

legimet, 2021-11-06:
JSMin is under the JSON license, which is considered [nonfree](https://www.gnu.org/licenses/license-list.en.html#JSON). Would it be possible to switch to a different JS minifier under a free license, such as [rjsmin](https://github.com/ndparker/rjsmin)? rjsmin is a replacement for jsmin.c written in Python.

jow-, 2021-11-07:
I do not consider the issue important enough to spend effort on it. If the “The Software shall be used for Good, not Evil.” license constraint makes it impossible to use jsmin in your porject, you can disable minification through menuconfig.

closed 2021-11-07 by jow-

legimet, 2021-11-27:
Hi, thank you for your response. I understand that most people probably don't care about this, but if I submitted a PR replacing it with a minifier under a free license, would that be accepted?

ysc3839, 2021-11-27:
Personally I suggest using Node.js based tools such as [UglifyJS](https://github.com/mishoo/UglifyJS). I think in the future, we will use more and more Web front-end tools. For example, we could use jsx to declare HTML elements instead of current `E('div')`.

jow-, 2021-11-27:
> but if I submitted a PR replacing it with a minifier under a free license, would that be accepted?

As long as it does not intorduce new build system prereqs it would be fine with me

> Personally I suggest using Node.js based tools such as UglifyJS.

I don't want the LuCI build to depend on node.js

> For example, we could use jsx to declare HTML elements instead of current E('div').

I don't want any additional frameworks in LuCI.

legimet, 2021-11-27:
I had suggested [rjsmin](https://github.com/ndparker/rjsmin), which is essentially a drop-in replacement for jsmin.c, except written in Python and released under the Apache License. According to the readme, "for simple integration you can just copy rjsmin.py into your project and use it." So I do not think there should be any issue.