openwrt/luci #5501: uHTTPd option "Do not follow symlinks outside document root" breaks luci

M95D, 2021-11-08:
To reproduce:
- Install luci-app-uhttpd
- Check option "Do not follow symlinks outside document root"
- Observe numerous Luci errors.

PS: It would help a lot debugging this kind of errors (and [that](https://github.com/openwrt/luci/issues/4555) kind of errors) if Luci error messages were more detailed and logged!

M95D, 2021-11-08:
Status / Firewall: Never finishes loading. Browser console shows periodic 404 errors for XHR POST https://<router-ip>/cgi-bin/cgi-exec?<some numbers>
Status / System log: Red message on top of the page: "Unable to load log data: Not Found"
System / Software: Red message on top: "NotFoundError".
Statistics / Graphs: Never finishes loading. Browser console shows error "Uncaught (in promise) NotFoundError: Not Found" and a couple of 404s.
Statistics / Setup: Never finishes loading. Browser console fills up several pages of 404s.

jow-, 2021-11-09:
Yep, could be a problem. I m not able to reproduce your issue with a default setup.

M95D, 2021-11-09:
I don't think it's necessary to rebuild the fw image to test this. Just remove the symlink and create a normal /var instead.

I'm rebuilding now an image without permanent /var, but it will take a while...

Tabiskabis, 2021-11-10:
When i activated "do not follow symlinks outside document root", it broke a few luci pages such as "Administration" and "Software" (can't recall the others, but these were annoying). Exactly the same issue.

jow-, 2021-11-11:
Ah yes, now I know why. It probably breaks the various cgi-io applets (cgi.-download, cgi-upload, cgi-exec) which symlink to /usr/libexec/cgi-io.

M95D, 2021-11-11:
Please have a look at that other error I mentioned:
Busybox with CONFIG_BUSYBOX_CONFIG_INSTALL_NO_USR will move /usr/bin/env to /bin/env and that breaks luci too.

jow-, 2021-11-11:
Yes, it will break LuCI since it expects /usr/bin/env to be available. No way around that

M95D, 2021-11-12:
Why not create /usr/bin/env symlink (if missing) at the end of package/busybox/install ?

closed 2024-12-04 by M95D