openwrt/luci #5517: Button to generate MAC

castillofrancodamian, 2021-11-14:
Not a problem, but I would like to see a button to generate a random MAC in the device or interface settings.

systemcrash, 2021-11-16:
Here u go

[image]

Che15ea, 2022-07-04:
Mind telling me where I can find this button?

Ansuel, 2022-07-04:
Mh? You mean a button to generate random Mac?

castillofrancodamian, 2022-07-04:
In KDE plasma there is a button to generate random MACs. I think it is a very useful function.

[image]

Che15ea, 2022-08-01:
> […](#)

Yes. Can we program a function that allow random/custom mac enabled by default. I mean to let the Openwrt forget about the original MAC address and never expose the original MAC address anytime anywhere. Broadcasting real MAC address is just too...big brother.

castillofrancodamian, 2022-08-01:
@Che15ea Someone already made a similar [proposal](https://github.com/openwrt/luci/pull/5896) of the function.

adelton, 2025-01-11:
With https://github.com/openwrt/luci/pull/5896 merged, can this issue be closed as resolved?

castillofrancodamian, 2025-01-11:
@adelton It's not the same. I thought it might be a good implementation of MAC generation on devices like WAN to get a different IP.

[image]

It would be nice if there was a button next to or below that section.

Neustradamus, 2025-01-12:
@adelton: Good ticket too, thanks!

Linked to:
- https://github.com/openwrt/luci/issues/5517
- https://github.com/openwrt/luci/issues/5819
- https://github.com/openwrt/luci/issues/5845
- https://github.com/openwrt/luci/pull/5896
- https://github.com/openwrt/luci/issues/7131