openwrt/luci #5523: "Official documentation" link points to obsoleted documentation

jelabarre59, 2021-11-16:
On the top-level "Documentation" page (https://github.com/openwrt/luci/wiki/Documentation) it points to the "official UCI-Documentation".   However this link points to the older, deprecated/read-only archive site (https://oldwiki.archive.openwrt.org/doc/uci).   Shouldn't this be pointing to something like https://openwrt.org/docs/start ?

jow-, 2021-11-16:
The new wiki does not have a suitable replacement for the old UCI reference pages. I see nothing in https://openwrt.org/docs/start that would be able to replace it.

bobafetthotmail, 2021-11-18:
In the new wiki that page is here https://openwrt.org/docs/guide-user/base-system/uci so you probably want to link that

jow-, 2021-11-18:
Ah, perfect - thanks!

closed 2021-11-18 by jow-

bobafetthotmail, 2021-11-18:
I have now fixed the link in the github documentation page so it points to the article in the new wiki