openwrt/luci #5530: Button in LuCi to install overlay packages defined in installed_packages.txt

shoulders, 2021-11-18:
It is now possible  with sysupgrade to get a list of your user installed packages (on the overlay) and have this put in `installed_packages.txt` which is preserved during a firmware upgrade.

However there appears to be no easy way to install these missing packages (for newbies) so what I purpose is putting a button on the `Backup / Flash Firmware` page to allow end-users with one click to install these missing packages (assumes working internet).

It would also be useful to have the option to create the `installed_packages.txt` and update it if required when generating a config backup

zorxd, 2021-11-18:
This should be handled by sysupgrade itself, not by LuCI.
It could start by a file which would contain a list of packages to install after a software update.
You typically don't want to reinstall all packages anyways. You only want a few ones, and let opkg handle the dependencies.

shoulders, 2021-11-19:
The button could run something like

```
# opkg update
# grep "\toverlay" /etc/backup/installed_packages.txt | cut -f1 | xargs -r opkg install
# rm /etc/backup/installed_packages.txt
# reboot
```
Taken from here https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg43129.html

zorxd, 2021-11-19:
well again, that would reinstall all packages, which is not what you typically want. Let say you want package A which depends on package B. After an update, it now depends on package C. Your script would install package B for no reason.

shoulders, 2021-11-19:
I completely understand, but i am just thinking of the hobbyist at home who struggles to keep his OpenWrt router up to date because of the hoops that have to be jumped through. A good chunk of users like me only have installed what they want and just want a simple update process.

You could add a list with checkboxes when the button is click which still utilises this file, however this would be a lot more complicated to add that a button :)

zorxd, 2021-11-19:
One of the problem is packages like libwolfssl4.7.0.66253b90. This is the package name. The version is 4.7.0-stable-2.
There could well be a different libwolfssl after the next update.

I still think a list of packages to (tentatively) reinstall after the software update would be a better idea, and it should be built-in to sysupgrade, not LuCI.