openwrt/luci #5535: Sqm settings off center in bootstrap

CaboBeach, 2021-11-21:
The sqm settings in bootstrap are 'off center' in Snapshots, settings themselves work fine and it's still readable, just wasn't sure if it was intended or not

OpenWrt SNAPSHOT r18153-617fcd4879 / LuCI Master git-21.324.29356-9cb6bde

[image]

[image]

[image]

[image]

hnyman, 2021-11-21:
I noticed the same yesterday with a new master build.
Possibly caused by the recent bootstrap CSS style changes by @jow- and the SQM app's style selections.

jow-, 2021-11-21:
Fixed with 4720a025e8 - the labels are still far too long imho - the left hand caption should be very short and concise, the extended field descriptions should be used for long explanations

commit 4720a025 (2021-11-21): luci-theme-bootstrap: fix cbi-value caption alignment

Since the removal of the clearfix styles, overlong captions will displace
subsequent input field rows.

Fix the issue by replace floating label + field margin with flexbox styles.

Fixes: #5535
Fixes: 8055acc9be ("luci-theme-bootstrap: overhaul styles")
closed 2021-11-21 by jow-